DarkFlippers/unleashed-firmware #100: Desktop percentage

Yoanndp, 2022-10-10:
### Describe the bug.

When you change the battery view, it doesn't change on the desktop. The view requires a reboot (Holding `Left` + `Back`) to get updated.

### Reproduction

1. Go to `Menu`
2. Select the `Settings` sub menu
3. From there, pick `Desktop` 
4. Finally, change the `Battery View` to whatever you want

### Logs

```Text
N/A
```

xMasterX, 2022-10-11:
Its not a bug, its how it was implemented, to avoid reading int memory every tick

In place where we updating screen we don't know when settings was updated, and we can't force update it right after change
It does not require reboot(but yes you can use it to speed up process), it will update automatically when battery percentage change or if you connect charger

Yoanndp, 2022-10-11:
Alright, it makes sense now. Thank you !

closed 2022-10-11 by Yoanndp