DarkFlippers/unleashed-firmware #96: Application browser as desktop preferred app

elrod16, 2022-10-10:
### Description of the feature you're suggesting.

The new ability to select a specific external app is nice, but the loss of the ability to use the app browser itself as a preferred app now is a little disappointing.

gteague, 2022-10-10:
you can still get to the browser with the down arrow. that's why i used the
2nd function of the left arrow for something other than apps. but now i use
it for video poker! :) /guy

elrod16, 2022-10-10:
Didn't mean to close.

I know about the browser but I preferred the ability to directly jump to the app browser with a left click. The normal browser takes a lot more than a simple click, unless I want to spam my favorites with a bunch of apps instead of my top used remotes and whatnot.

gteague, 2022-10-10:
to me the useless button is up arrow because i'm not going to use any of
the 3 options. is there any reason some favorites couldn't replace that
menu (user optional, of course) or added to that menu? also, the right
arrow is pretty useless as well as the status could be added to the /about/
selection i'd think. oh well, i'm not a programmer so i have no idea what's
possible. i like my flipper so much i'm not going to complain about it. /guy

elrod16, 2022-10-10:
It is in the settings menu under Passport already.

elrod16, 2022-10-11:
It might be different if folders were allowed to be pinned in favorites

gteague, 2022-10-11:
ah! right you are. so it's already redundant. i haven't looked in that
passport option since i got mine and went through all the settings. /guy

closed 2023-06-01 by elrod16