DarkFlippers/unleashed-firmware #101: Flipper Bluetooth capability seems under utilized

casper89d, 2022-10-11:
The Flipper's Bluetooth module seems mostly forgotten or over-looked. Any chance for increasing the flippers Bluetooth options/capability like pairing with multiple devices?

emory, 2022-10-18:
a basic "proximity sensor" function seems like a logical `hello world` for steering attention into Bluetooth

ankris812, 2022-10-18:
im trying to adapt a example for a stm32wb55dk (a discovery kit board) but its getting imposible.

ankris812, 2022-10-18:
> a basic "proximity sensor" function seems like a logical `hello world` for steering attention into Bluetooth

good idea

marckranat, 2022-10-18:
how about a brute force for bluetooth as there are few combinations, you could have it start checking all the common defaults, then just bruteforce.

ankris812, 2022-10-18:
> how about a brute force for bluetooth as there are few combinations, you could have it start checking all the common defaults, then just bruteforce.
well , i prefer starting with something basic first. im trying to make scanning work. then i can work on the bruteforce

jfinstrom, 2022-12-13:
It is my understanding that the BT radio stack is stripped due to memory constraints. There has been several btle discussions in the forums where the bluetooth is mostly there as a serial interface.   That said it seems the marauder bluetooth functionality is mostly ignored.  People are building esp32 boards they may as well use that radio stack for wifi/bt/ble

ankris812, 2022-12-13:
I can confirm you that right now, full stack and firmware both fit at the
same time

cyberpz, 2023-01-22:
AirTag capability might be a possible one, not my idea.

ankris812, 2023-01-23:
Only if It can be enabled and disabled. Ask roguemaster what happened
because he enabled It when It wasnt posible disable it

sethrice, 2023-04-16:
Having spent hours futzing over the years with Kismet on Kali, for Wifi, SDR, Bluetooth, and zwave radio devices. I was thinking about this, and wondering if there were any hardware limitations on the flipper for bluetooth. I expect we could do a lot with the bluetooth radio in there, so long as you aren't connecting from your phone via the app.

sysfu, 2023-04-24:
> It is my understanding that the BT radio stack is stripped due to memory constraints. There has been several btle discussions in the forums where the bluetooth is mostly there as a serial interface. [...]

So far I've had zero luck sniffing bluetooth packets using Wifi Dev Board with Flipper GPIO => [ESP32] WiFi Marauder app. It runs the sniffbt command but it logs no packets or devices. Going to open an issue for that shortly.

At the very least it would be nice to have a raw bluetooth packet sniffing mode that would write a pcap file to the SDcard.

storm1er, 2023-04-25:
Just for reference as I was looking for BT stuff, I think #346 is relevant here :)

Leptopt1los, 2024-01-06:
The flipper cannot act as a host in a ble connection. acting as a slave, it cannot be connected to more than one device at the same time, like any other ble slave devices, for example, headphones

closed 2024-01-06 by Leptopt1los

ankris812, 2024-01-23:
> The flipper cannot act as a host in a ble connection. acting as a slave, it cannot be connected to more than one device at the same time, like any other ble slave devices, for example, headphones

you can, but not with current stack