DarkFlippers/unleashed-firmware #106: NFC-V support

TheKogoro, 2022-10-13:
### Description of the feature you're suggesting.

Many NFC system using NVC-V keycard (ISO 15693) that isnt supported by flipper zero actually. NFC-V is main used by most of Skydata system (Parking lot, Ski station, and many other system...). 

Actually, NFC-V is "readable" but cannot do nothing with it on flipper. It only showing a number without any possibility

nemanjan00, 2022-11-22:
That is due to limitation of NFC chipset used in flipper

vikwin, 2022-12-06:
> That is due to limitation of NFC chipset used in flipper

That's not true. It's just not implemented on official firmware yet. See this pull request, someone already implemented a working draft with further NFC-V / ISO 15693 features:
https://github.com/flipperdevices/flipperzero-firmware/pull/1991

nemanjan00, 2022-12-12:
As I said, lib and chip is not designed for anything other than reading of NFC-V...

Anything else is usage of chip outside of spec and hack

nemanjan00, 2022-12-13:
I have to correct myself. Apparently, there is raw mode which can be used to interact with ISO15693 cards. 

Thanks to @bettse for correcting me

Flaneo, 2023-01-13:
I would LOVE to see the support for NFC-V Tags as soon as possible :)

Enteee, 2023-01-14:
well, there is this: https://github.com/flipperdevices/flipperzero-firmware/pull/1991

FoxLy1312, 2023-01-14:
> well, there is this: flipperdevices/flipperzero-firmware#1991

Awesome, thanks for your Work :D
Make my day <3

nemanjan00, 2023-01-19:
PR is not draft anymore. It is pending review now

Maybe time to merge?

https://github.com/flipperdevices/flipperzero-firmware/pull/2316

amec0e, 2023-01-19:
> PR is not draft anymore. It is pending review now
> 
> Maybe time to merge?
> 
> [flipperdevices/flipperzero-firmware#2316](https://github.com/flipperdevices/flipperzero-firmware/pull/2316)

Merging is blocked on that PR, until it is merged it is not ready.

RogueMaster, 2023-01-19:
You can ask the developer @g3gg0, this has been ready for a bit

amec0e, 2023-01-19:
Assuming OFW don't want to make any additional changes

g3gg0, 2023-01-19:
In this case the code is not 100% clean as it still directly access rfal* functions from some other layer.
And there are for sure also some other quirks which are not that great.

Flipper zero crew are taking things serious.
They have set rules and seem to follow them, which is a good practice.
So not having merged the PR already is absolutely okay and I am fine with that.

The underlying PulseReader/DigitalSequence PR was said to get merged soon, so I think
I have reached the expected code quality for that part. And for the NfcV stuff I hope to get some
feedback, but I think this will take until their NFC API rework goes live.

Thats the "does it follow my arhcitecture" point of view.

From a "stability" point of view however, I consider the NfcV stuff stable enough for a merge.
Haven't seen any crashes, hickups or nonfunc situations.
It was also successfully used by some other dev as underlying layer for picopass emulation.

So you can decide if it matches your criteria.

In any case, I would really love to see other people joining the work on it and e.g. file PRs to improve the NfcV code.
There was a question if I could add writing memory payload to another card, maybe add some other card types etc.
And also the rfal or generally adoption to the new NFC API are things, help would be appreciated :)

Regards,
g3gg0

xMasterX, 2023-01-19:
> You can ask the developer @g3gg0, this has been ready for a bit

We will wait for official team to review that changes
Or if author wants to send copy of that PR to our repo directly we can do our basic review (we have no dedicated nfc developers)
and merge it

xMasterX, 2023-01-19:
> In this case the code is not 100% clean as it still directly access rfal* functions from some other layer. And there are for sure also some other quirks which are not that great.
> 
> Flipper zero crew are taking things serious. They have set rules and seem to follow them, which is a good practice. So not having merged the PR already is absolutely okay and I am fine with that.
> 
> The underlying PulseReader/DigitalSequence PR was said to get merged soon, so I think I have reached the expected code quality for that part. And for the NfcV stuff I hope to get some feedback, but I think this will take until their NFC API rework goes live.
> 
> Thats the "does it follow my architecture" point of view.
> 
> From a "stability" point of view however, I consider the NfcV stuff stable enough for a merge. Haven't seen any crashes, hickups or nonfunc situations. It was also successfully used by some other dev as underlying layer for picopass emulation.
> 
> So you can decide if it matches your criteria.
> 
> In any case, I would really love to see other people joining the work on it and e.g. file PRs to improve the NfcV code. There was a question if I could add writing memory payload to another card, maybe add some other card types etc. And also the rfal or generally adoption to the new NFC API are things, help would be appreciated :)
> 
> Regards, g3gg0

Yes, you are right, but sometimes reviews from official team takes longer than expected 
So if possible can you make a duplicate of that PR but to us direclty? if not possible do you agree if I merge it by committing it manually, but a bit later

g3gg0, 2023-01-19:
Oh, thanks for that offer.
Right now I am working on so many topics, be it flipper related (https://github.com/flipperdevices/flipperzero-firmware/pull/2311) or some other stuff (http://g3gg0.de) or even daily life, that I tend to get lost somewhere inbetween.
Opening another PR here would most likely cause a cerebral stack overflow.

I hope thats okay for you :)

xMasterX, 2023-01-20:
> Oh, thanks for that offer. Right now I am working on so many topics, be it flipper related ([flipperdevices/flipperzero-firmware#2311](https://github.com/flipperdevices/flipperzero-firmware/pull/2311)) or some other stuff (http://g3gg0.de) or even daily life, that I tend to get lost somewhere inbetween. Opening another PR here would most likely cause a cerebral stack overflow.
> 
> I hope thats okay for you :)

Sure, I understand
I think for now better solution will be to wait official team review, also im seeing that other users found issues and you fixed them already

TheKogoro, 2023-01-21:
I also See on discord that NFC-V (Proprietary) will surely never work with flipper zero (Skidata for example)

> You can ask the developer @g3gg0, this has been ready for a bit

g3gg0, 2023-01-21:
> I also See on discord that NFC-V (Proprietary) will surely never work with flipper zero (Skidata for example) 🤔
can you explain that?

TheKogoro, 2023-01-21:
> > I also See on discord that NFC-V (Proprietary) will surely never work with flipper zero (Skidata for example) 🤔
> 
> can you explain that?

Yeah, i have ask If Skidata keycard (NFC-V) Iso15693 will be working with flipper zero and it seem to be Not possible without the datasheet or key. But i am Not an expert so i dont know really what is needed and somes other people say it never be possible.

If having physically the card can help to Make it working i can Always sens it one.

g3gg0, 2023-01-21:
if you aren't an expert, then please correct your statement as it is simply incorrect.

NfcV is possible and working. Even emulating is working. e.g. NXP SLIX* cards can already be emulated.

Regarding skidata, I have no clue what the reader is doing, but the latest PR adds a sniff feature.
One could sniff the stuff the reader is sending to figure out what is happening.
If you have any reliable information source, please link it.

TheKogoro, 2023-01-21:
> if you aren't an expert, then please correct your statement as it is simply incorrect.
>
> NfcV is possible and working. Even emulating is working. e.g. NXP SLIX* cards can already be emulated. [...]

I know i am Not an expert but is What people answering  me that isnt be possible on official flipper zero discord under NFC Chanel. (Thanks for correct that by telling me this 👍🏻)

For the Sniffing feature, is with the "detect Reader"  function ? If is that, the nfc coder of Skidata detect Flipper zero (as non useable card) but flipper zero doesnt detect the Skidata coder station.

Also sorry for misunderstand and bad Spelling, i Speak french and in english + technical speaking is relatively hard ^^

g3gg0, 2023-01-21:
This would require you to manually compile and flash the branch of my pull request.

ranqizhu, 2023-02-24:
@g3gg0 thanks for the work! I was testing it a bit from using the rougemaster release. It reads/saves the NXP-SLIX2 card well however emulation not working. My iphone NFC reader app is not detecting it but it recognizes the original card well.

g3gg0, 2023-02-24:
yeah same here. when reading using a phone, the reception quality is quite bad.
not sure why, but even with a scope i can see that the pulses are incorrect sometimes.
this makes me suspicious regarding the antenna tunings for the NFC chip or the transparent mode of that chip.

with a proxmark or a toniebox however the range for the flipper is comparable to a nfc sticker

M3NVU5, 2023-02-27:
Bonjour,
Est-il juste possible de la cloner (copier et coller sur une autre)?
Si oui, grace à quel Update?
Merci par avance ;-)

g3gg0, 2023-02-27:
french is not the most universal language in the world. please use english.

M3NVU5, 2023-02-27:
> le français n'est pas la langue la plus universelle au monde. veuillez utiliser l'anglais.

Good morning,
Is it just possible to clone it (copy and paste on another)?
If so, which update?
Thanks in advance ;-)

g3gg0, 2023-02-27:
you cannot just copy NfcV cards. 
there are however "magic cards" which allow UID modification.
you can however, depending on the exact type, emulate it on flipper.

Desko21, 2023-03-07:
just tried this using the rougemaster release, good job @g3ggo!

TheKogoro, 2023-03-08:
> just tried this using the rougemaster release, good job @g3gg0! Tested with ISO15693, SLIX and SLIX-L and worked like a charm!

I hope it will be soon in Unleashed too 🙏🏻

xMasterX, 2023-05-03:
> Hello @g3gg0
> 
> I tested reading an NFC-V card via unleashed and the flipper was only able to give me the UID.
> 
> By testing your NFC-V reader module via RogueMaster, I got a lot more information and was able to emulate the card too!
> 
> Unfortunately, the reader does not react by offering him the emulator card from the FZ.
> 
> I therefore wonder about the emulation which seems incorrect. I have to try to copy the FZ card to a real SLI/SLIX UID card that can be found here: https://lab401.com/fr/products/icode-sli-slix-compatible-uid-modifiable?_pos=10&_sid=0a2e4b59f&_ss=r
> 
> If you want to take a look at the copy of the NFC-V card (copy-nfc-v-iso15963.nfc) with which I tested it is available here: https://file.io/AeR1zNkIFpis
> 
> Thank you for advancing the reading of NFC-V

There is n NFC V support in unleashed, what you tested?

Misstmourt, 2023-05-03:
> > Hello @g3gg0
> > I tested reading an NFC-V card via unleashed and the flipper was only able to give me the UID.
> > By testing your NFC-V reader module via RogueMaster, I got a lot more information and was able to emulate the card too!
> > Unfortunately, the reader does not react by offering him the emulator card from the FZ.
> > I therefore wonder about the emulation which seems incorrect. I have to try to copy the FZ card to a real SLI/SLIX UID card that can be found here: https://lab401.com/fr/products/icode-sli-slix-compatible-uid-modifiable?_pos=10&_sid=0a2e4b59f&_ss=r
> > If you want to take a look at the copy of the NFC-V card (copy-nfc-v-iso15963.nfc) with which I tested it is available here: https://file.io/AeR1zNkIFpis
> > Thank you for advancing the reading of NFC-V
> 
> There is no NFC V support in unleashed, what you tested?

I test on RogueMaster (on unleashed, NFC-V dot not work).  I think @g3gg0 has made further progress on some of these branches (I found recent commits on this thread)

TheKogoro, 2023-05-03:
> I test on RogueMaster (on unleashed, NFC-V dot not work). I think @g3gg0 has made further progress on some of these branches (I found recent commits on this thread)

Same problems to mer with Skidata card (i tested on Xtreme firmware). It read the card but cannot read fully (it say 34 blocks, truncated to 32 blocks)
[image]

[image]

xMasterX, 2023-05-03:
Hey guys, why you discussing it in Unleashed firmware repo? one talking about RM, other about Xtreme, this is wrong place for such discussion

xMasterX, 2023-05-03:
You should test NFC V not in Xtreme or RM

But in original PR in official firmware, since there was some big changes recently, by nvx
And discuss issues in official firmware PR that implements nfc v

if you "testing" it on forks they might be using outdated source code merged from times when it was firstly made and PRs was in draft state

MadjidZ, 2023-05-07:
You're right @xMasterX.

FYI I tested g3gg0 branch ISO15693, and I encountered the same issue. I compared your code with that of g3gg0 , there is no diff for read/save/emulate NFCV. 

We'll have to wait a while longer before we see support for NFCV on the FZ ;-)

g3gg0, 2023-05-07:
> We'll have to wait a while longer before we see support for NFCV on the FZ ;-)

how did you come to that conclusion?

ClaraCrazy, 2023-05-07:
> if you "testing" it on forks they might be using outdated source code merged from times when it was firstly made and PRs was in draft state

dont worry, we always update remotes we pull from. But ofc testing should be done on the ofw PR to see if it works or not.

MadjidZ, 2023-05-07:
By testing with your branch which seems the most successful for reading and emulating NFCV cards based on ISO15693 specifications. 

I have found that much more information can be read (previously only the UUID). Unfortunately the emulation does not seem to work. The NFCV reader does not react when passing an FZ emulating a previously saved NFCV card.

I think that like other NFC types, some cards will be readable and can be copied via the FZ and others will not.

g3gg0, 2023-05-07:
first of all - "emulating ISO15693" is like "emulating ISO14443".
you cannot "generally emulate" those technologies as they are the underlying protocol for a lot of cards.
you can only "emulate NXP SLIX-L" oder "SLIX2" etc which use the protocol.

the only generic thing is "emulate the NfcV UID" which is as pointless as for NfcA. only totally broken designs can get tricked by it.

so please get a bit more specific about the card technology and the readers you are experimenting with.

g3gg0, 2023-05-08:
the file he posted was "generic ISO15693" with just an UID and zeroes in the memory blocks.
so nothing that would help you as there is no feature that will rewrite magic generic ISO15693 cards.

niggowai, 2023-05-08:
OK thx anyway

TheKogoro, 2023-05-15:
Hello, If it can help, Here is two type of 15693 card. One is an ski card (iso 15693 SLIX S) and the other a non coded (fresh card, never used/coded) an iso 15693 from Skidata.

[ISO 15693 SLX S Ski card](https://file.io/rmXiMV1NXQ2O)
[ISO 15693 Skidata Card](https://file.io/Rc0UaSmYdwJv)

niggowai, 2023-05-16:
Finally found the Commands to copy the chinese Ones in the [proxmark repo](https://github.com/Proxmark/proxmark3/pull/842/files#diff-1a6224a88680a1a080999c64307568643212c0333712f5757db4ac4dd02f183aR1629-R1645):

According to that, two simple write commands do the trick on a rewriteable card.

0x02 21 38 {last 4 bytes of uid}
0x02 21 39 {first 4 bytes of uid}

The order of these commands isnt important, but the uid needs to be written in little Endian

e.g.:

UID E0 04 01 08 43 07 88 FC

would result in the Commands:

-> 0x02 21 38 FC 88 07 43
-> 0x02 21 38 08 01 04 E0

I Can confirm it working with that [card from lab401](https://lab401.com/products/icode-sli-slix-compatible-uid-modifiable) and my Android Phone via [NFC-Tools App](https://play.google.com/store/apps/details?id=com.wakdev.wdnfc)

closed 2023-07-21 by xMasterX

henrygab, 2023-11-07:
```
-> 0x02 21 38 FC 88 07 43
-> 0x02 21 38 08 01 04 E0
           ^^---- I presume this should be 39 ?
```

niggowai, 2023-11-08:
yes sure :sweat_smile: