DarkFlippers/unleashed-firmware #112: Dummy Mode shortcut to GAMES list

Stefo0, 2022-10-14:
### Description of the feature you're suggesting.

Hi,
If I activate the Dummy Mode on the Flipper Zero the OK button opens directly the Snake game. 
Is it possible to change it and show the game list instead of opening Snake Game? Otherwise is not possible to play any other game if Dummy Mode is activated.

If it's not possible, how can I set another game as default game in Dummy Mode?

Thanks..

TQMatvey, 2022-10-15:
well this should be implemented on the official side, as dummy mode is the official featur that they should develop, currently you cannot set any other game in dummy mode..

TQMatvey, 2022-10-15:
> Thanks..

Thats actually a very interesting topic, i am trying to make it to work also right now, the problem is that if i open games folder on dummy mode, you can still go out to other FAPs which is very bad :<

Stefo0, 2022-10-15:
Is possible to create a selection option (like the Desktop/Primary Favorite App...it could be named Dummy Favorite App) where the user can decide what to open in Dummy Mode? He can select one of the games available or the game list.

TQMatvey, 2022-10-15:
possible but difficult

RogueMaster, 2022-10-26:
This should help fix this... or at least improve on it... #137

xMasterX, 2022-10-30:
This was implemented in PR #137 from RogueMaster but not yet user customisable, this will be for next time

closed 2022-10-30 by xMasterX