DarkFlippers/unleashed-firmware #123: Remove Payment card NFC restrictions

ALD53105, 2022-10-19:
### Describe the enhancement you're suggesting.

It would be nice to have the restrictions on saving and emulating payment card NFC’s removed. I completely understand why this was removed from the OFW, but it would be nice to have added back in Unleashed. I understand the implications of this request, but please remember it can already be done just by flashing an older version of the firmware.

ankris812, 2022-10-19:
you can only emulate uid, thats not enough to emulate a card, so its a useless feature, it was removed to svae space, because it did nothing

xMasterX, 2022-10-20:
Flipper was never been able to emulate full EMV card, this feature was UID emulation which is useless, only might be useful if you using bank card as access card to office, but it very very rare case
So, for what usage you need saving and UID emulation of bank cards?

xMasterX, 2022-10-21:
UID Emulation and saving function is back, might be useful with some access control system that uses EMV cards as access cards
But if reader uses EMV for auth it will not work at all!

closed 2022-10-21 by xMasterX