DarkFlippers/unleashed-firmware #129: unlshd-006 seems to have issues emulating NFC

vashmalachi, 2022-10-23:
### Describe the bug.

After updating to `unlshd-006`, I attempted to emulate an Amiibo file that was known to work and no device would read it. Tested with another Flipper on stock firmware and a Pixel 3. Both devices could read that tag before updating when emulated by my Unleashed Flipper. All saved NFC files seem affected.

### Reproduction

Update to `unlshd-006` using web updater.
Navigate to NFC > Saved > Any NFC file > Emulate.
Hold to NFC reader.
Get no output from reader, as if no tag was presented.

xMasterX, 2022-10-24:
Tested with NTAG215, emulation works fine, send your "broken" file here
Looks like problem is on your side, or with that particular file

vashmalachi, 2022-10-24:
I can send a few, if you'd like but they are just the ones from the amiibo db on the awesome flipper github. I also tried with some random saved NFCs that could be read before the update. After downgrading to the previous version it works again. Would you still like me to send some files?

xMasterX, 2022-10-24:
send files that don’t work, or issue will be closed because bug can’t be reproduced

vashmalachi, 2022-10-24:
To further clarify, it wasn't just one file. All files I tried to emulate failed to be read.

vashmalachi, 2022-10-24:
Here are some that I tried:
[NFCs.zip](https://github.com/DarkFlippers/unleashed-firmware/files/9848391/NFCs.zip)

xMasterX, 2022-10-24:
I tested your files, emulation works fine, so the problem on your side, and can’t be replicated

All files worked fine, can be read with proxmark and with smartphone

closed 2022-10-24 by xMasterX