DarkFlippers/unleashed-firmware #131: Frequency analyser shortcut in bruteforce app

BrunoKlu, 2022-10-24:
### Description of the feature you're suggesting.

When performing a subghz brutforce attack it would be nice to have a shortcut that quickly enable frequency analysyser in case a target is comming, you can quickly switch to frequency analyser and know if you choose the right frequency for the bruteforce.

xMasterX, 2022-10-24:
Frequency analyzer works in distances like 5-10cm 

not like 100 meters

So this is not possible

closed 2022-10-24 by xMasterX

BrunoKlu, 2022-10-24:
By the way Is there any sources telling how the frequency analyser is working ?

BrunoKlu, 2022-10-24:
Is it because the reciever sensivity is very bad compare to the transmitter output power ?

xMasterX, 2022-10-24:
https://github.com/DarkFlippers/unleashed-firmware/blob/dev/applications/main/subghz/helpers/subghz_frequency_analyzer_worker.c