DarkFlippers/unleashed-firmware #133: Sort files in browser feature

vivi7, 2022-10-24:
I have thousands of file in infrared folder for example by brand, but they are order random or by added date, I want to order alphabetically.

FalsePhilosopher, 2022-10-24:
sudo fatsort -n /dev/sdb1

xMasterX, 2022-10-27:
Also solution for Windows:

https://www.anerty.net/software/file/DriveSort/

RogueMaster, 2022-10-30:
@xMasterX I have tested https://github.com/RogueMaster/flipperzero-firmware-wPlugins/pull/389 to be working well

Supertera, 2022-11-22:
This is not work with 128 Gb SD-card, cause it is formatted (by flipper) in exFAT file system. And fatsort don't work with it probably the same reason.

Supertera, 2022-11-22:
My wish, about it... may be someone will adding sorting utility to the "application"

RogueMaster, 2022-11-22:
> My wish, about it... may be someone will adding sorting utility to the "application"

Just build RogueMaster#389 into your FW

Supertera, 2022-11-22:
> Just build [RogueMaster#389](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/pull/389) into your FW

Awessome! Please, add this in the future release of unleashed firmware...

xMasterX, 2023-02-13:
Now implemented in firmware

closed 2023-02-13 by xMasterX