DarkFlippers/unleashed-firmware #134: Shutdown Idle

Sorrow81, 2022-10-25:
### Description of the feature you're suggesting.

Hello,

I regularly forget to stop my tool, I saw that on RogueMaster an automatic stop can be programmed. Is it possible to integrate this with the Unleashed version?

https://github.com/RogueMaster/flipperzero-firmware-wPlugins/blob/420/applications/settings/power_settings_app/scenes/power_settings_scene_shutdown_idle.c

Thanks.

RogueMaster, 2022-10-26:
This is the real source https://github.com/flipperdevices/flipperzero-firmware/pull/1647

xMasterX, 2022-10-30:
Will be added after modifications made to check if any activity is happening, in current state it will power off regardless of user actions

AlexQuaken, 2023-04-25:
What about integration, quite important functionality ?

Leptopt1los, 2024-01-06:
Will be merged after merging https://github.com/flipperdevices/flipperzero-firmware/pull/1647 to OFW

xMasterX, 2025-02-21:
Done

closed 2025-02-21 by xMasterX