DarkFlippers/unleashed-firmware #136: Add 330 MHz in setting_user

LowSkillDeveloper, 2022-10-26:
### Describe the enhancement you're suggesting.

Might be worth adding "Frequency: 330000000" to setting_user by default? On this frequency works same remote controls, various gates and barriers also used it. And if this frequency is added, the flipper can catch them and decode.

[image]

Or is there any reason for the absence of this frequency?

portman56, 2022-10-26:
Can we not add it ourselfs? Edit the settings_user file?

xMasterX, 2022-10-26:
Added to config and frequency analyzer, thanks!

closed 2022-10-26 by xMasterX