DarkFlippers/unleashed-firmware #138: Add time delay for favorite emulation

ximik666, 2022-10-27:
### Description of the feature you're suggesting.

Add a time delay to the setting (5,10,15,30 seconds). This delay will disable key emulation from favorites after a specified period of time.
For example, the user has selected 5 seconds. It selects the desired key from the favorites and emulates it. After a specified period of time, the emulation is turned off and there is no need to force it to turn off by pressing the Back button.
Thank you.

xMasterX, 2022-10-30:
Is pressing back button really that hard? I don’t understand purpose of that feature

Leptopt1los, 2024-01-06:
We closed it as not planned
Reopen the issue if you can describe a real scenario for using the requested feature

closed 2024-01-06 by Leptopt1los