DarkFlippers/unleashed-firmware #146: Sub-GHz BFT Mitto 433MHz not reading correctly the .sub files

gitlaman, 2022-11-01:
### Describe the bug.

As the title says i have problem with the creation and also with old files that are created through Sub-GHz manual added BFT Mitto remotes. At first all was working as it should, i created a file through manual add and when i run the file i was able to send the signal and everything, after 6 to 10 executions the file or the plugin or the device stuck and now i formatted the sd card, factory restarted the device, even DFU recovery installation with the original flipper firmware and nothing works, when i execute the .sub file that i created and every new file even and old (the Cnt value is always Cnt:0000 and also the MF: Unknown) i tried every possible thing that i can think off but nothing works and this is only for the Mitto, i tried the RM firmware and i have the same result, i dont know what im doing wrong or where did i messed up. Here are the before and after screenshots to get what i mean. 

[image]
[image]

Filetype: Flipper SubGhz Key File
Version: 1
Frequency: 433920000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: KeeLoq
Bit: 64
Key: A1 66 30 01 10 E2 D0 04
Seed: 92 8B 77 B9

Please help. 
Thanks in advance

### Reproduction

Im not really sure because i cant fix the problem.
i captured 4 raw signals from my remote, sent them to decrypt the seed, after i got the seed i entered everything and it was not working due to my error, then the user sent me .sub file with settings according to my gate and i tried it 6 to 10 times and it was working, after that i pressed my original remote 6 times, the next thing i was unable to execute .sub files that are new created even not the old ones, tried other sub-ghz like FAAC SLH to add manually just to see if the values show up and everything is ok. So somehow the BTF plugin is stuck and wont show the MF and Cnt and when i press send all the values are the same and not changing.

xMasterX, 2022-11-01:
I can reproduce bug, will be fixed soon

xMasterX, 2022-11-01:
Fixed, fix will be published in next release, thanks for report

closed 2022-11-01 by xMasterX