DarkFlippers/unleashed-firmware #152: [SubGHz] Reading 64bit KeeLoq seem to be highly unreliable

ppseprus, 2022-11-06:
### Describe the bug.

I have a couple of test keyfobs that I use to make sure things work fine.

I was testing [7175459](https://github.com/DarkFlippers/unleashed-firmware/commit/7175459bcd734c8174957ab7485cf3df4209a8a3) when I realised that pushing the same button on the same keyfob, out of 60 tries, about 58-59 I get an _unknown_ manufacturer, while sometimes I get something random.

After multiple rounds of 60 signals, I basically saw all kinds of manufacturers: BFT, Aprimatic_TR, JCM_Tech, Sommer(fsk476), Cenmax, Partisan_RX, Mutanco_Mutancode, Teco, etc.

Here are some keys from the same button of the same test keyfob:
```
Protocol: KeeLoq
Bit: 64
Key: 4D E6 58 4F 4A 82 60 08
Manufacture: Faraon

Protocol: KeeLoq
Bit: 64
Key: C7 0F E3 46 4A 82 60 08
Manufacture: Tomahawk_Z,X_3-5

Key: 32 FA 36 FE 4A 82 60 08
Manufacture: Teco

Protocol: KeeLoq
Bit: 64
Key: DA 87 1F AF 4A 82 60 08
Manufacture: Mutanco_Mutancode

Protocol: KeeLoq
Bit: 64
Key: 13 9A A6 AD 4A 82 60 08
Manufacture: Partisan_RX

Protocol: KeeLoq
Bit: 64
Key: 63 5E 45 28 4A 82 60 08
Manufacture: Sommer(fsk476)
```

Unfortunately, Im not an expert at C or C++, so...
- Is this due to changes in unleashed, or should I open this bug report at the official repository?
- How is MF resolved from these signals? Shouldn't the same key always produce the same MF?
- Is this a cosmetics issue, where the name is extracted unreliable, but other parts are intact OR does this mean that some other functionalities of SubGHz reading is always affected somehow?

### Reproduction

1. Switch on, Open SubGHz Read
2. Select the frequency and modulation of your 64bit KeyLoq target (for me, 433.92 MHz, AM270)
3. Push one button on your keyfob
4. Repeat Step 3. until memory is full

**Expected Behaviour**
MF is consistently identified (or consistently _unknown_, depending on exact manufacturer)

**Actual Behaviour**
Out of 60 tries, 1-2 are some random (always different) manufacturers, while about 58-59 times I get _unknown_

xMasterX, 2022-11-08:
Hello, this is not a bug, that’s how keeloq decoding works, unknown or random manufacturer means flipper has no manufacturer code for your remote, random manufacturers appear because of how algorithm is designed, 

If you have genius idea how to detect with 100% correct results go ahead and describe all in detail, me and official firmware developers will be interested in it

closed 2022-11-08 by xMasterX

ppseprus, 2022-11-08:
Oh! I did not know that about KeeLoq! Thanks for explaining!

>If you have genius idea how to detect with 100% correct results go ahead and describe all in detail, me and official firmware developers will be interested in it

I wish! 😅