DarkFlippers/unleashed-firmware #155: Lib <lib/toolbox/value_index.h>/<lib/toolbox/value_index.c> works bad for APP plugins

hrundeel, 2022-11-07:
### Describe the bug.

While I try to build my fap (applications/plugins) it returns error: "app won't run. Unresolved symbols: {'value_index_uint32', 'value_index_float'}". And app don't work. 

If I build my own app on apllications/main - all ok.
If I copy this functions direct to my source - it's seems all ok.

Don't know how it fix for myself for flawless build.

### Reproduction

1. Project in applications/plugins.
2. Add to source files <#include <lib/toolbox/value_index.h> and use functions like uint8_t value_index_uint32(*), uint8_t value_index_float(*), uint8_t value_index_bool(*).

### Target

Fix the problem to build correctly.

xMasterX, 2022-11-10:
Thanks, Added to API, pull latest dev, all should work now

closed 2022-11-10 by xMasterX

hrundeel, 2022-11-10:
Thanks, but this don't help. 
Building is ok (without any errors/warnings now), but after uploading .fap to "Sd Card/Misc" and trying to run - it not work with message "Error Load failed. Found unsatisfied imports".
Don't know how to do.

xMasterX, 2022-11-10:
> Thanks, but this don't help. Building is ok (without any errors/warnings now), but after uploading .fap to "Sd Card/Misc" and trying to run - it not work with message "Error Load failed. Found unsatisfied imports". Don't know how to do.

Do you know that you need to actually build and flash firmware from our dev branch?
not just compile app and throw it on old release that doesn’t have that functions?

xMasterX, 2022-11-10:
Run this with flipper connected `./fbt COMPACT=1 DEBUG=0 FORCE=1 flash_usb`
And after that, try launching your app

hrundeel, 2022-11-10:
> Do you know that you need to actually build and flash firmware from our dev branch?
> not just compile app and throw it on old release that doesn’t have that functions?

Shure. Deleted dist/build and build again. .fap build ok, but not working with his error.

xMasterX, 2022-11-10:
> > Do you know that you need to actually build and flash firmware from our dev branch?
> > not just compile app and throw it on old release that doesn’t have that functions?
> 
> Shure. Deleted dist/build and build again. .fap build ok, but not working with his error.

You need to flash that firmware from dist, not just build it

xMasterX, 2022-11-10:
Use command for fbt that I suggested

hrundeel, 2022-11-10:
> Run this with flipper connected `./fbt COMPACT=1 DEBUG=0 FORCE=1 flash_usb`
> And after that, try launching your app

Thanks! Now it's help. I'm updating from qFlipper before.