DarkFlippers/unleashed-firmware #173: NRF24L01 support sniffing

CasualArclamp, 2022-11-20:
### Description of the feature you're suggesting.

It would be awesome if https://github.com/RogueMaster/flipperzero-firmware-wPlugins/blob/420/documentation/NRF24.md
could be merged. NRF24 support would be a great addition to this firmware Microsoft key jack sniffing and decrypting the key stokes is relatively trivial due to Microsoft's botched implementation. sammy shows it here https://youtu.be/WqkmGG0biXc

TQMatvey, 2022-11-20:
the app is already included, check Applications -> GPIO -> [NRF24] Mouse jacker and [NRF24] Sniffer

xMasterX, 2022-11-20:
Looks like you haven’t even looked into our readme and have not tried to install firmware 
Read Matvey’s answer

closed 2022-11-20 by xMasterX