DarkFlippers/unleashed-firmware #184: New protocol for applications>plugins>weather_station

LY2NEO, 2022-11-24:
### Describe the enhancement you're suggesting.

I already update my local repository with new weather_station protocol (HG06061A-DCF-TX sensor EU) tested ant it is working with my station. But I am totally green with GITHUB and at this moment do not understand how to do pull request, please help :)

xMasterX, 2022-11-24:
You need to create fork from our repo, than apply all changes in your fork, and after you can send a PR

Read these instructions:

https://www.freecodecamp.org/news/how-to-make-your-first-pull-request-on-github-3/

https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request

xMasterX, 2022-11-24:
Or alternatively you can send all modified files here in this issue, also with raw recording of your weather station transmitter
to allow me to test it

I can add changes manually and add info that you made it, in changelog and in commits

LY2NEO, 2022-11-25:
Thanks for info!
This time just uploading here and going to watch github tutorial :)

[WS_AURIOL.zip](https://github.com/DarkFlippers/unleashed-firmware/files/10090086/WS_AURIOL.zip)

xMasterX, 2022-11-25:
I looked into code, seems fine, ill add this into dev branch soon
Also, do you want me to make PR to official repo with your code? 
Ill write that you are author and i'm helping with PR

LY2NEO, 2022-11-26:
Thanks a lot, for helping me. Yes, please make PR for me at this time.

xMasterX, 2022-11-26:
> Thanks a lot, for helping me. Yes, please make PR for me at this time.

I made PR for you in official repo:
https://github.com/flipperdevices/flipperzero-firmware/pull/2056

Also will commit it to our dev branch too, thanks!

closed 2022-11-27 by LY2NEO