DarkFlippers/unleashed-firmware #186: General imrovements from Model-7

Supertera, 2022-11-26:
### Description of the feature you're suggesting.

[image]
](url)
I think the developers of the flipper are familiar with a device called "model-7". This device has a lot of manufacturing codes and protocols that are not implemented in the flipper. Question - why? There is no way to contact one of the developers of model-7? To be honest, I thought that the developers of the flipper and the model-7 were the same people. I'm wrong? Tell me, why reinvent the wheel and add one protocol and a manufacturing code to the flipper, when you could combine the functionality with model-7 and make a truly amazing tool! My advice is to find the right person, especially with such funding...

RocketGod-git, 2022-11-26:
https://autodecoders.com/product/j1-self-learning-remote-control-with-frequency-adjustable-new-model/

Here is the link to buy one. Very expensive and doesn't look like it has anything Flipper doesn't.

Supertera, 2022-11-26:
I was able to buy such a device for $200. Believe me, no offense to the flipper, it can work with many rolling codes, opens almost all barriers. At the moment, the flipper can work with 1/4 of the codes contained in the model-7. Unfortunately, the model-7 project is abandoned, unlike the flipper, so I hope that the flipper will someday catch up with the model-7 and maybe overtake it. But again, why add a teaspoon (one code at a time) to the new firmware when you can find the developers of model-7, buy the firmware from them and improve the flipper at once. It's worth it !!!

Supertera, 2022-11-26:
Take a look to long "Model-7" list... and this is not full list...

Allmatic B.RO OVER
Allmatic B.RO STAR
Allmatic B.RO WN
Allmatic TECH 3
AN-MOTORS Черные и серебряные кнопки
APRIMATIC TR2/TR4
BENINCA T4WV
BENINCA CUPIDO 4
BENINCA TO GO 2/4WV
BYTEC/BAISHENG
Bytec TB-4
Baisheng RC4-T15
BFT MITTO 2/4, 2N/4N 
CAME TAM 432
CAME TOP 432/434
CAME TOP 432/434 EE
CAME ATOMO AT 01/02/04
CAME TOP 862NA (868Mhz)
COMUNELLO KEEP 2/4
COMUNELLO KEEP VICTOR 2/4
Doorhan Transmitter 4/2/D2
Doorhan RCBLACK-4
DTM NEO
DEA Mio tR 2/4
DEA 273 (серые кнопки, gray buttons)
DITEC GOL4
DITEC GOL4C (красный светодиод, red LED)
DITEC BIXLS2
FAAC XT4 433RC
FAAC TM1/2/3-DS
FAAC 868SLH
FAAC XT2/XT4 868SLH – with external pc postprocessing
Marantec D222/D224/214 (868Mhz)
Marantec Digital 302/313/304 (868Mhz)
NORMSTAHL/CRAWFORD RCU433-4K, FHS10-01/02, EA433-4KS/2KS, EA433-2KM, N002800, T40-1/1M, T40-4/4M
MOOVO MT4/MT4G/MT4V
Nice Flo 4/2/1
Nice Flor-s (-m) 4/2/1
Nice Very VE
Nice Very VR
Nice Smilo 2/4, blue/black
Nice Era Flor
Nice Era Inti (INTI1x/INTI2x)
Nice Era One (ON1E…ON9E)
NERO Radio 8101-½/4 (434Mhz)
NERO Radio 8101-2М (434Mhz)
NOVOFERM Novotron 302/502
ROGER H80 TX22
ROGER H80 TX12/14
GENIUS BRAVO
GENIUS ECHO TX4
GSN TXRC09
GSN TXRC10
GSN TxRC09TB
GSN TxRC10TB
GSN TR-300S
GSN TR-500F
ELMES UMB
ELMES DWB and compatible
HORMANN HSM4, HSE2, HSP4 (868Mhz)
Keeloq systems and much more....

mayday77, 2022-11-26:
Mate. It's cost 100$ I'm Moscow )

RocketGod-git, 2022-11-26:
Link? All I see is discontinued in RU.

mayday77, 2022-11-26:
> Link? All I see is discontinued in RU. 

[image] this is fresh version, with nfc + more sweet drops.

xMasterX, 2022-11-26:
> ### Description of the feature you're suggesting.
> [image]
> 
> I think the developers of the flipper are familiar with a device called "model-7". This device has a lot of manufacturing codes and protocols that are not implemented in the flipper. Question - why? There is no way to contact one of the developers of model-7? To be honest, I thought that the developers of the flipper and the model-7 were the same people. I'm wrong? Tell me, why reinvent the wheel and add one protocol and a manufacturing code to the flipper firmware release, when you could combine the functionality with model-7 and make a truly amazing tool! My advice is to find the right person, especially with such funding...
> 
> ### Anything else?
> _No response_

Who are you? Yes you are wrong, developers of the flipper and the model-7 NOT!!!!! the same people
Combine functionality? How do you want to do it? Maybe you have source code of model 7 fw? share it then

xMasterX, 2022-11-26:
Also if you want to talk with developers of the flipper, then you are in wrong repo

Here you can tell them your genius idea:
https://github.com/flipperdevices/flipperzero-firmware

closed 2022-11-26 by xMasterX

Eng1n33r, 2023-09-29:
Barely done with this device. Sources are available.

Supertera, 2023-09-30:
Nice to hear! I believed that you could do this. This will bring the device to a decent level!