DarkFlippers/unleashed-firmware #190: Sub-Ghz Universal remote

PharoahCoder, 2022-11-27:
### Description of the feature you're suggesting.

Hi,

Has anyone thought of creating a universal remote just like the infrared but for sub-ghz frequencies for things like restaurant pagers and gates/barriers. each device type can have a group to access the universal remote for it for example infrared has acs, tvs, projectors. Sub-ghz universal remote can have a barriers section, pager section and so on. Maybe even an option specifically for restaurant pagers that allows the restaurant id to be bruteforced. I really think this would be a good addition to the firmware.
I would do this myself but my experience with C is very limited.
Let me know what you guys think.

Inspiration:
[Xenobyte's creation of the t119 bruteforcer](https://twitter.com/xenobyte_/status/1554222170888339456?lang=en)

Contact me at  ak@yousefak.com for any questions
PS: For educational purposes

askoriy, 2022-11-28:
I think it's not possible. Unlike TVs and ACs that have the same and well-known codes for the model, every gate or restaurant pager has its own code, that could be found only by sniffing or bruteforce.

RocketGod-git, 2022-11-28:
This feature already exists.

[image]
[image]
[image]

PharoahCoder, 2022-11-28:
> I think it's not possible. Unlike TVs and ACs that have the same and well-known codes for the model, every gate or restaurant pager has its own code, that could be found only by sniffing or bruteforce.

Thats why im suggesting there should be a built in bruteforce mode for each brand that doesnt require exporting analyzing and manually creating raw bruteforce sub files.

PharoahCoder, 2022-11-28:
> This feature already exists.
> 
> [image] [image] [image]

I meant something to bruteforce though. Also that doesn't work for me it says "please configure map" even though they are already there.

PharoahCoder, 2022-11-28:
> You can use whatever .sub files you want. Read the readme to learn how to make your own map files.

Yeah but that means you cant do it straight from the flipper without emulating it. You would have to listen then export it to create a bruteforced version. A better way would be to bruteforce straight from the device with preconfigured files just like the acs.ir and fans.ir

matthewholt1, 2022-11-30:
> Thats why im suggesting there should be a built in bruteforce mode for each brand that doesnt require exporting analyzing and manually creating raw bruteforce sub files.

Each brand of TV? Sometimes not even the same TV manufacturer uses the same frequencies.

xMasterX, 2022-12-02:
> > This feature already exists.
> > [image] [image] [image]
> 
> I meant something to bruteforce though. Also that doesn't work for me it says "please configure map" even though they are already there.

https://github.com/DarkFlippers/unleashed-firmware/blob/dev/documentation/SubGHzRemotePlugin.md

closed 2022-12-15 by xMasterX