DarkFlippers/unleashed-firmware #192: For the devs, unsupported protocol

killer4nl, 2022-12-02:
### Description of the feature you're suggesting.

Hi guys,

Here are 2 unsupported protocol files > https://github.com/killer4nl/Captured-files-Flipper

### Anything else?

For the devs

assasinfil, 2023-02-10:
Need more info about the key. (Photo, chip name, labels,...)

killer4nl, 2023-02-12:
I don't really have much info besides photo's and the flipper dump

https://gyazo.com/5e2ea4281ccf58386cd07a99112c393a
https://gyazo.com/c5547da7af0deb07736ce3bce49fbece
https://gyazo.com/b0e569db815acb782d032f2e34970ca5

It's a tag to open doors, it works with a receiver on the wall

xMasterX, 2023-02-13:
Please post this in OFW repo since we have no experience with rfid protocols and can't help with implementing it at the moment

supaeasy, 2023-07-18:
Looks like an interflex key to me. I wonder how you even got these files. Mine doesnt read at all.