DarkFlippers/unleashed-firmware #193: Magspoof LF RFID implementation

zacharyweiss, 2022-12-02:
### Description of the feature you're suggesting.

Howdy! Some of you may already be familiar with Samy Kamkar's "Magspoof" POC ([website](https://samy.pl/magspoof/) / [GH repo](https://github.com/samyk/magspoof)). For those unfamiliar, the TL;DR is one is able to convince magstripe readers a card has been swiped by modulating an electromagnet to reproduce inserting and removing a card.

It appears previously, members of the Flipper community discussed an implementation that uses the Flipper's LF RFID coil to achieve this ([forum post](https://forum.flipperzero.one/t/feature-request-magspoof/256)) and it was [implemented in a fork of the official FW](https://github.com/dunaevai135/flipperzero-firmware/tree/feature/nomodule/applications/magspoof) but never merged with main. This seems to be a great candidate to include with Unleased; there's a number of swipe access cards I'd love to try emulating with a Flipper Magspoof app.

zacharyweiss, 2022-12-03:
From a bit more digging, seems there's additional spots this has been worked on:
-[Russian Flipper hackathon which included a magspoof project](https://habr.com/ru/company/flipperdevices/blog/595325/)
- [An old commit to Unleashed](https://github.com/DarkFlippers/unleashed-firmware/commit/9e1dc86560d11d3a89fff76df303ba75b720399b)
- [And a more recent attempt by the wetox team](https://github.com/wetox-team/flipperzero-firmware/tree/ezhevita/magspoof/applications/plugins/magspoof)

Leone-Montana, 2022-12-06:
https://discord.com/channels/937479784148115456/1026058078921166939/1041159629020860506

jluo98, 2022-12-07:
> https://discord.com/channels/937479784148115456/1026058078921166939/1041159629020860506

Thanks for the link! I managed to compile it for the latest firmware. Do you happen to know what type/format of file this reads? I have a couple of raw magnetic stripe info ready, just not sure how to format those so the app can read them.

zacharyweiss, 2022-12-08:
> Thanks for the link! I managed to compile it for the latest firmware. Do you happen to know what type/format of file this reads? I have a couple of raw magnetic stripe info ready, just not sure how [...]

From a very quick skim of that code, looks like the file loading is handled in magspoof_device.c. Seems to expect a file in a "magspoof" folder, with ".magspoof" as the extension, and a header that reads "Flipper Magspoof device". Not yet sure how it expects the rest of the data in the file to be formatted. 

How did you compile for the most recent FW? I've yet to set up a dev environment for Flipper; any guides you followed?

jluo98, 2022-12-08:
> How did you compile for the most recent FW? I've yet to set up a dev environment for Flipper; any guides you followed?

The compiling guide is [here](https://github.com/flipperdevices/flipperzero-firmware/blob/dev/documentation/AppsOnSDCard.md) in the official repo. I did see an error while compiling and the app does run but crashes sometimes. The code probably needs some fixing.

zacharyweiss, 2022-12-08:
Appreciate the link! I gave that a go; cloned unleashed, set up fbt, modified the application.fam manifest to designate it as FlipperAppType.EXTERNAL, and built fap_magspoof without errors, but when copied onto the Flipper it still gives me a preload failed error claiming the API version is mismatched, despite my Flipper running unlshd-018. Was there something else you needed to modify, or specify when building the app, for the API versioning to be happy?

zacharyweiss, 2022-12-08:
@jluo98 To your earlier question on data format, it's using flipper_format. While I'm still pretty unfamiliar with the these libraries / tools, it seems a data file (magspoof/*.magspoof) would read as follows.

Filetype: Flipper Magspoof device
Version: 1
Data: _data_

Where _data_ is just a string. Seeing as most magstripe readers are just HIDs, perhaps just try whatever the plaintext output of a swipe is? I went to test if this file would load, but attempting the "Saved cards" action causes a repeatable crash for me, failing some furi_assert. I'll need to find that assert, but if you're able to open the "Saved cards" option on yours, do let me know if that format of file is opened/parsed properly.

jluo98, 2022-12-09:
@zacharyweiss Very interesting findings. Using the saved card feature did bring up the file browser, but selecting the test file I prepared caused a crash. 

AFAIK flipper does not have any onboard sensors to read a magnetic stripes, only emulation is possible with RFID hardware. The read function is probably written for an external module.

zacharyweiss, 2022-12-09:
RX is definitely external; seems that's what they're importing the UART library for.

TX is internal. Seems the few confusing comments I was seeing re: H-bridge / motor driver etc were just directly from the original magspoof project (and not indicative of an external module for TX). Most of the code in the read_card scene seems to be directly copied from Samy Kamkar's repo, and just modified slightly to work with the Flipper API and relevant I/O. Leaves me wondering if it wouldn't be easier to start from scratch with Samy's code as the base, rather than fixing this half-finished port.

zacharyweiss, 2022-12-12:
So, interestingly it seems the [original hackathon project intended for external TX](https://github.com/flipperdevices/flipper-hackathon-moscow/issues/30) based on this old schematic I found. It seems future passes / iterations have been done with the attempt of using the RFID coil, but it's hard for me to evaluate their effectiveness as the RFID API remains opaque to me. 

It seems there's no reason it wouldn't be able to work on just the RFID coil; [this guy managed to simplify the hardware to something analogous](https://www.idogendel.com/en/archives/507). Presumably, if we just have an easy way to emit simple OOK signals from the RFID antenna we'd be golden. I've been struggling to understand the RFID API / relevant pins / etc though. Are there any resources or folks who'd be able to offer insight? The FURI HAL code itself isn't well enough commented for me to follow.

zacharyweiss, 2022-12-18:
Have been documenting further iterations / attempts [in this Discord thread](https://discord.com/channels/937479784148115456/1052419406896054352).

zacharyweiss, 2022-12-27:
A few quick updates: 
- I've been able to pick up a signal on a janky antenna I've made for my oscilloscope. It's weak, but at least indicates I'm able to pull true coil
- Currently rewriting all the boilerplate / scenes / file code as it's exceptionally broken, and easier to just start from scratch
- If the signal remains weak after some more messing with it, next ideas (in no particular order) are adding some kind of subcarrier freq, trying the timer system instead of direct-writing the pins, trying the NFC coil additionally/instead

zacharyweiss, 2022-12-28:
See here for my [WIP ground-up rewrite](https://github.com/zacharyweiss/magspoof_flipper). I'm using the README as very coarse notes on what remains to be done — anyone is welcome to contribute!

jluo98, 2022-12-28:
This is awesome! Thanks so much for working on this!

skynet01, 2023-01-16:
Any updates on this :) Or for sure we will need an add-on antenna? Btw thanks for taking time and writing WIP in that readme file

zacharyweiss, 2023-01-17:
I'll be building in the option for TX over RFID or an external GPIO connected coil. RFID does seem to work, just weaker. Been busy with other work so haven't had much time to dedicate to this the past week.

skynet01, 2023-01-17:
Woo! Set up a buy me a beer link, I am sure some of us would be happy to contribute and motivate your efforts :)

zacharyweiss, 2023-01-22:
A few updates; reworked the file format and added a variety of emulation config options — you can now TX over RFID or GPIO, change the clock and interpacket durations, and select the track to emulate. I've also included a sample file in `assets` (add it to the `mag` folder at the Flipper SD's root) for anyone testing it. Reverse track emulation still yet to be implemented, and much cleanup (especially of saved info displayed, and the add manually process) + tuning + testing remains. Eventually I'll start exploring RX options as well, but I anticipate that'll be a large project.

Also added a buy me a coffee link at the bottom of the README! Please don't feel compelled to donate — I'm doing this solely for fun / my own sake — but I won't turn down a drink :)

skynet01, 2023-01-23:
Thanks for the updates! Looks like your latest version is now included in the latest RM Firmware (the patreon one) for those who want to try it out without compiling it manually.

zacharyweiss, 2023-01-23:
Cool! It's still likely too unfinished for anyone other than fellow developers to mess with yet, but if that gets other folks interested in digging through the code with me -- all the better.

Re: RX methods, does the Flipper companion app allow "sideloaded" community tools? What's the process by which MFkey32 got included?

skynet01, 2023-01-23:
Not sure actually, but there is a thread with no answers on flipper forum here: https://forum.flipperzero.one/t/what-is-the-hub-and-how-can-the-community-interact-with-it/10588 maybe if we all ask them more it will get their attention? 

Another idea for getting card info is we can maybe import saved NFC cards (the bank cards that got scanned that have a chip)?

zacharyweiss, 2023-01-23:
NFC-scanned EMV cards don't expose the same data as is on the magstripe (card number is preserved across the two, of course, but you lack critical info like CVV; which one should also note is different from the CVV2 that is physically written on cards). At the end of the day though my primary focus remains access cards; abuse potential/misuse I feel is a larger concern with payment-card-specific functionality.

RogueMaster, 2023-01-23:
> Cool! It's still likely too unfinished for anyone other than fellow developers to mess with yet, but if that gets other folks interested in digging through the code with me -- all the better. One thing though: would you mind including it as a submodule rather than just copying the files? It'll keep things up to date for you, and make it easier for folks to find their way back to my repo.

Hi, I don't mind manually updating as needed. I prefer it since I also make changes to the code. I don't add any applications as submodules. Hopefully this gets other people digging into it with you, or supporting you further to keep working on it 😄

zacharyweiss, 2023-01-23:
Understandable. Would likely be better to fork it in that case, and include your own fork as a submodule, but as long as you're including credit and keeping the license, no worries.

skynet01, 2023-01-23:
> IIRC NFC-scanned EMV cards don't expose the same data as is on the magstripe (card number is preserved across the two, of course, but you lack critical info like CVV; which one should also note is different from the CVV2 that is physically written on cards). At the end of the day though my primary focus remains access cards; abuse potential/misuse I feel is a larger concern with payment-card-specific functionality.

You might be right, I was just thinking of a faster way to adding my cards to flipper, I know that those saved NFC files are more of a reference since very little is included and its different for each card. Still saves typing that big number  :) Thanks for considering it.

rhysperry111, 2023-04-04:
Hey- will be getting a Flipper Zero soon and I'm pretty interested in this feature, so I'd love to help in some way.

I have access to a [USB magnetic stripe reader](https://www.rfideas.com/products/readers/magnetic-stripe/pcswipe), which seems to output card data as ASCII for RX. Not sure how exactly I can get the raw card data from that tough.

Also, another thing that I'm not sure how easy it would be to implement is the direction the emulated card moves. Many room key readers require the card to be inserted and then removed in succession in order for the lock to open.

rhysperry111, 2023-04-12:
Finally got my flipper, would be happy to help get this moving again <3

closed 2024-01-06 by Leptopt1los

FatherDivine, 2024-03-04:
> Finally got my flipper, would be happy to help get this moving again <3

any updates?

RogueMaster, 2024-03-04:
> any updates?

I nominate you as tribute

FatherDivine, 2024-03-04:
> > any updates?
> 
> I nominate you as tribute

I just ordered a [USB mag reader](https://www.amazon.com/gp/product/B01DUB4GVO/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) that will be delivered in a couple of hours the same day. Already have some old (hotel) magstripes to test. What are the immediate needs for this project? Also, does the internal (or R.L. Multi Pass) currently work with card readers that you traditionally insert a magstriped card inside? 

If a solution is needed for processing multiple raw card data at once, I can create a PowerShell(or pwsh) script that processes the raw data that would work for *nix and mac OSX. Could turn into an Android/IOS app once no large issues with the magnetic reader's driver interfacing with Android/IOS... This would be using an OTG cable or USB-A to USB-C adapter to connect to the phone... I also run full Linux on one of my unlocked AOSP-based devices. I figure the bigger picture is having portability (hence having the phone read the data via a small headphone jack/lightning-based Square reader). Being able to read full magstripes and write as well as emulate "on the fly" and in the field.

FatherDivine, 2024-03-05:
> > any updates?
> 
> I nominate you as tribute

Here is my contribution and update to this project:
**[Magspoof Flipper Card Organizer](https://github.com/FatherDivine/f0-dev/tree/main/magspoof_flipper_cardorganizer)**

it's a WIP, and if it's slow at work today I'll finish the code and documentation today as well.

Basically it's a PowerShell module (that can be ran as individual scripts/functions) that organizes magspoof data. Since it's a module, you can call it anytime you open a PowerShell session and type: 

`invoke-magspoofcardorganizer`

That will load it into a mode of detecting the usb reader input to organize (WIP, not working right yet).

if you wish to load a dump file (from scanning strips to a text file and saving as whatever you wish... I use '.dump') then you would use this cmdlet:

`invoke-magspoofcardorganizer  -FileName "C:\path\to\mags.dump"`

or the alias

`magorganizer -File "C:\path\to\mags.dump"`

The above 2 cmdlets currently just outputp on the screen. 

But again I should have all the rest coded in today. I coded what's there now while falling asleep after a long day.

P.S. @RogueMaster  thanks for keeping RM as good as it is! That's my firmware of choice.

zacharyweiss, 2024-03-31:
Hey folks! Sorry for largely abandoning the project, work got notably busy and progress petered out on internal coil TX; could get it to produce a weak field, but never strong enough to get picked up by a reader w/o an external GPIO coil. Glad to see y'all have been making strides in my absence and continuing to improve the FAP :)