DarkFlippers/unleashed-firmware #200: A/C remote database is much less flexible than in OFW

makkarpov, 2022-12-07:
### Describe the enhancement you're suggesting.

Currently, A/C remote database in this FW differs drastically from official FW in the way of organization. Official FW encodes complete actions (i.e. "turn off", "turn on & set to medium cooling") and this FW encodes button presses (i.e. "temp+", "temp-", "toggle power").

It appears that many A/C's encode all parameters in the IR packet, which makes differential actions impossible. This is why official database has such organizations. For example, official FW works well on Sinclair A/C which is hanging near me, and this FW does not work at all - it even fails to turn it off.

The situation is worsened by the fact that action set is hardcoded into scene, and you cannot just swap databases. I'm opening this as an ehnacement request because I don't know exactly the right way to resolve that. Obviously, there were some reasons to design A/C remote database like this. Probably there are ACs which work well with this "button press encoding". Maybe some switch in settings? Or another button in a scene which toggles two databases?

xMasterX, 2022-12-10:
We looking how we can make our database compatible with different layout, when we come with solution it will be released

Leptopt1los, 2024-01-06:
Now we use ofw's layout for all ir assets

closed 2024-01-06 by Leptopt1los