DarkFlippers/unleashed-firmware #202: Desktop battery view issue

Flip-dex, 2022-12-08:
### Describe the bug.

After changing the selected battery display type in the settings, the newly chosen option does not take effect immediately. In fact it seems to only update the battery display type several minutes later or if the battery level itself changes invoking the display to update the style. 

### Reproduction

Go to settings -> desktop -> battery view

Change battery view type

Go back to desktop and witness the same battery view as before changing for several minutes.

H3rrOberst, 2022-12-09:
Thats also in the OFW.

Az-Neter, 2022-12-09:
I thought you were supposed* to reset for the change to take effect?