DarkFlippers/unleashed-firmware #205: Documentation incorrect

31337-4554551n, 2022-12-13:
### Describe the bug.

Documentation says to put the file in the "playlist" folder
The folder it looks for is "subplaylist"

### Reproduction

create a playlist folder
Add a playlist
open playlist app
See it defaulting to a different directory
Move file to subplaylist folder
open app
see it in the list

UberGuidoZ, 2022-12-19:
Could you link to the documentation? There are different utilities "playlist" and "subplaylist" depending on the application and firmware. Thank you.

31337-4554551n, 2022-12-20:
Yes, this was the one. Guess I can't tell the difference between what's official and what isn't yet. I followed the link from your front page where it talks about the play list plugin by clicking on the link next to it. 
[image]