DarkFlippers/unleashed-firmware #206: U2F over NFC

TheXHalferX, 2022-12-14:
### Description of the feature you're suggesting.

U2F is super handy feature of Flipper Zero, but it lacks NFC functionality in stock firmware. Adding such feature (U2F over NFC) will be very cool.

MattCheetham, 2022-12-29:
There is a draft PR over on the official firmware but it does seem to have gone quiet. https://github.com/flipperdevices/flipperzero-firmware/pull/1755

dedsecproyt, 2023-06-06:
From my build i have error 

LoadAppManifest, line 33, in file "/home/dedsec/Flipper/unleashed-firmware/scripts/fbt_tools/fbt_apps.py"
	SDKCHK	firmware/targets/f7/api_symbols.csv
API version 28.2 is up to date
	APPS	build/f7-firmware-C/applications/applications.c
	CC	applications/main/u2f/scenes/u2f_scene_error.c
	CC	applications/main/u2f/scenes/u2f_scene_main.c
	CC	applications/main/u2f/views/u2f_view.c
	CC	applications/main/u2f/u2f.c
applications/main/u2f/views/u2f_view.c: In function 'u2f_view_draw_callback':
applications/main/u2f/views/u2f_view.c:17:38: error: 'I_Drive_112x35' undeclared (first use in this function)
   17 |     canvas_draw_icon(canvas, 8, 14, &I_Drive_112x35);
      |                                      ^~~~~~~~~~~~~~
applications/main/u2f/views/u2f_view.c:17:38: note: each undeclared identifier is reported only once for each function it appears in
applications/main/u2f/views/u2f_view.c:21:43: error: 'I_Connect_me_62x31' undeclared (first use in this function)
   21 |         canvas_draw_icon(canvas, 22, 15, &I_Connect_me_62x31);
      |                                           ^~~~~~~~~~~~~~~~~~
applications/main/u2f/views/u2f_view.c:25:43: error: 'I_Connected_62x31' undeclared (first use in this function)
   25 |         canvas_draw_icon(canvas, 22, 15, &I_Connected_62x31);
      |                                           ^~~~~~~~~~~~~~~~~
applications/main/u2f/views/u2f_view.c:29:43: error: 'I_Auth_62x31' undeclared (first use in this function)
   29 |         canvas_draw_icon(canvas, 22, 15, &I_Auth_62x31);
      |                                           ^~~~~~~~~~~~
applications/main/u2f/views/u2f_view.c:41:43: error: 'I_Error_62x31' undeclared (first use in this function)
   41 |         canvas_draw_icon(canvas, 22, 15, &I_Error_62x31);
      |                                           ^~~~~~~~~~~~~
applications/main/u2f/views/u2f_view.c: In function 'u2f_view_set_ok_callback':
applications/main/u2f/views/u2f_view.c:93:10: error: macro "with_view_model" requires 4 arguments, but only 2 given
   93 |         });
      |          ^
In file included from applications/main/u2f/views/u2f_view.h:3,
                 from applications/main/u2f/views/u2f_view.c:1:
applications/services/gui/view.h:231: note: macro "with_view_model" defined here
  231 | #define with_view_model(view, type, code, update) \
      | 
	CC	applications/main/u2f/u2f_app.c
applications/main/u2f/views/u2f_view.c:87:5: error: 'with_view_model' undeclared (first use in this function)
   87 |     with_view_model(
      |     ^~~~~~~~~~~~~~~
applications/main/u2f/views/u2f_view.c:84:75: error: unused parameter 'context' [-Werror=unused-parameter]
   84 | _set_ok_callback(U2fView* u2f, U2fOkCallback callback, void* context) {
      |                                                        ~~~~~~^~~~~~~

applications/main/u2f/views/u2f_view.c: In function 'u2f_view_set_state':
applications/main/u2f/views/u2f_view.c:101:10: error: macro "with_view_model" requires 4 arguments, but only 2 given
  101 |         });
      |          ^
In file included from applications/main/u2f/views/u2f_view.h:3,
                 from applications/main/u2f/views/u2f_view.c:1:
applications/services/gui/view.h:231: note: macro "with_view_model" defined here
  231 | #define with_view_model(view, type, code, update) \
      | 
applications/main/u2f/scenes/u2f_scene_error.c: In function 'u2f_scene_error_on_enter':
applications/main/u2f/scenes/u2f_scene_error.c:16:53: error: 'I_SDQuestion_35x43' undeclared (first use in this function)
   16 |         widget_add_icon_element(app->widget, 0, 0, &I_SDQuestion_35x43);
      |                                                     ^~~~~~~~~~~~~~~~~~
applications/main/u2f/scenes/u2f_scene_error.c:16:53: note: each undeclared identifier is reported only once for each function it appears in
applications/main/u2f/scenes/u2f_scene_error.c:28:54: error: 'I_ActiveConnection_50x64' undeclared (first use in this function)
   28 |         widget_add_icon_element(app->widget, 78, 0, &I_ActiveConnection_50x64);
      |                                                      ^~~~~~~~~~~~~~~~~~~~~~~~
scons: *** [build/f7-firmware-C/applications/main/u2f/scenes/u2f_scene_error.o] Error 1
applications/main/u2f/views/u2f_view.c:97:5: error: 'with_view_model' undeclared (first use in this function)
   97 |     with_view_model(
      |     ^~~~~~~~~~~~~~~
applications/main/u2f/views/u2f_view.c:96:34: error: unused parameter 'u2f' [-Werror=unused-parameter]
   96 | void u2f_view_set_state(U2fView* u2f, U2fViewMsg msg) {
      |                         ~~~~~~~~~^~~
applications/main/u2f/views/u2f_view.c:96:50: error: unused parameter 'msg' [-Werror=unused-parameter]
   96 | void u2f_view_set_state(U2fView* u2f, U2fViewMsg msg) {
      |                                       ~~~~~~~~~~~^~~
cc1: all warnings being treated as errors
scons: *** [build/f7-firmware-C/applications/main/u2f/views/u2f_view.o] Error 1
applications/main/u2f/u2f.c:10:10: fatal error: toolbox/hmac_sha256.h: No such file or directory
   10 | #include "toolbox/hmac_sha256.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
scons: *** [build/f7-firmware-C/applications/main/u2f/u2f.o] Error 1

********** FBT ERRORS **********
build/f7-firmware-C/applications/main/u2f/scenes/u2f_scene_error.o: Error 1
build/f7-firmware-C/applications/main/u2f/views/u2f_view.o: Error 1
build/f7-firmware-C/applications/main/u2f/u2f.o: Error 1