DarkFlippers/unleashed-firmware #207: Bravia TV's no longer compatible in IR universal remote

fOmey, 2022-12-15:
### Describe the bug.

Since the infared assets update (commit: 6f9f0dbe6ee236378dcb7bd302e2f63ed33905fb) Sony Bravia TV's no longer appear to be supported in the universal remote.

I've tested with power toggle, mute & volume controls.

### Reproduction

1. Open IR universal remote.
2. Point flipper at Sony Bravia TV.
3. Toggle a button.

GideonWolfe, 2022-12-19:
I've also been noticing the Universal Remote working less after switching to Unleashed.  Others have [noticed the same](https://www.reddit.com/r/flipperzero/comments/zohsfn/universal_remote_not_effective/)

UberGuidoZ, 2022-12-19:
Interesting. Likely because something was changed in the way things were handled by Official. And since Unleashed is built off the dev base, something may need to be adjusted.

@fOmey Do you have a capture of the actual remote for comparison? I want to make sure the codes are still in the Universal file. For example, here's the POWER on a Sony Bravia XBR TV:
```
Filetype: IR signals file
Version: 1
# 
name: Power
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 15 00 00 00
```
It does exist in the tv.ir Universal file, but I have confirmed it's not working on the POWER. Speaking to Amec0e now about it so we can figure out where things stopped working. Thank you.

fOmey, 2022-12-27:
> Interesting. Likely because something was changed in the way things were handled by Official. And since Unleashed is built off the dev base, something may need to be adjusted.
>
> @fOmey Do you have a [...]

```
Filetype: IR signals file
Version: 1
# 
name: POWER
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 15 00 00 00
# 
name: VOL+
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 12 00 00 00
# 
name: VOL-
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 13 00 00 00
# 
name: CH+
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 8958 4449 510 4475 515 4444 515 2213 508 4477 513 2215 516 2212 509 2219 512 2217 514 2214 517 2211 540 2214 517 2211 510 4449 510 2218 513 4472 507 2220 511 30572 8960 2218 513 87698 8966 2211 510 87701 8963 2214 568
# 
name: CH-
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 8956 4451 508 2220 511 2217 514 4470 510 4449 510 2218 544 2211 510 2218 513 2215 516 2212 509 2220 511 2217 514 2214 517 2211 510 2245 517 4441 508 2220 511 35049 8961 2215 516 87696 8959 2217 514 87698 8956 2220 511 87701 8964 2213 508
# 
name: UP_Arrow
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 74 00 00 00
# 
name: Down_Arrow
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 75 00 00 00
# 
name: Left_Arrow
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 34 00 00 00
# 
name: Right_Arrow
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 33 00 00 00
# 
name: Enter
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 65 00 00 00
# 
name: Return
type: parsed
protocol: SIRC15
address: 97 00 00 00
command: 23 00 00 00
# 
name: Input
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 25 00 00 00
# 
name: MUTE
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 14 00 00 00
```

bunk3r, 2022-12-30:
Don't know why, but on dev-215 the Universal Remote standby command works again on my Bravia, but not the others (vol, mute, etc..)

xMasterX, 2022-12-30:
> Don't know why, but on dev-215 the Universal Remote standby command works again on my Bravia, but not the others (vol, mute, etc..)

We fixed power button, but other buttons for sony, using SIRC protocol doesn’t work unless they captured in RAW

ToXiCRaTt, 2023-01-07:
Same here no response with universal remote.

UberGuidoZ, 2023-01-07:
> Same here no response with universal remote.

For what model TV? We’ve tested on a handful that we’re not working before, but thanks to xMasterX idea, SIRC is now working. (Was an issue introduced from Official.)

xMasterX, 2023-01-19:
Fixed in latest release

closed 2023-01-19 by xMasterX