DarkFlippers/unleashed-firmware #212: Guide: How to use custom layouts in badUSB?

TheT3ngu, 2022-12-18:
### Description of the feature you're suggesting.

The [Keyboard layout file generator](https://github.com/dummy-decoy/flipperzero_badusb_kl) provides us with the possibility to create custom keyboard layouts for Flipper`s badUSB, and many of these layouts are preinstalled as .kl-files with the latest release.
But: How do we use a generated layout on the Flipper in badUSB? Does there need to be some kind of command in each ducky script that tells the Flipper what layout to use for the attack, or how do we tell the Flipper which layout to use? 
A guide on how to use custom layouts within for the badUSB-module would be a nice complement to the guide section.

Admonstrator, 2023-01-15:
I would guess that you can just throw them into `/ext/badusb/layouts`
(Or BadUSB/Layouts if you access your SD-Card directly)

TheT3ngu, 2023-01-22:
thought so too, but unfortunately, after deleting everything from /layouts but my local lang (de-DE) badUSB is still using default american layout when executing scripts.

Admonstrator, 2023-01-22:
Afaik you choose the layout to use before you start your badusb script. It can be choosen by selecting the options in the badusb app.

TheT3ngu, 2023-01-26:
You are right! For anyone who is as braindead as me, the process is:
1. select your script
2. DON'T hit run, but press < - Button and select layout
3. Hit run

closed 2023-01-26 by TheT3ngu