DarkFlippers/unleashed-firmware #224: [NFC] unlshd-022 firmware "Can't parse data from app" error when reading NFCs and "Can not parse file" when trying to open saved NFCs

V-kond, 2022-12-21:
### Describe the bug.

Since I have updated from unlshd-018 to unlshd-022 I noticed I am unable to utilize any newly saved NFC cards.
I am gettin the error message "Can't parse data from app" after reading a card. After saving, the .NFC file seems okay, but then there is the error message "Can not parse file" when I rty to emulate it/rub it in app.
Problem is consistent for any kind of NFC type I tried.

Would appreciate any kind of solution.
THX

### Reproduction

1. Open NFC.
2. Read any kind of NFC card for the "Can't parse data from app" error message.
3. Save NFC card data.
4. Re-open saved .NFC to emulate (Run in App) for the "Can not parse file"  error message.

GueroHabanero, 2022-12-21:
Thats odd, i have not gotten an error like this maybe your sd card for flipper? Try rebooting your flipper

V-kond, 2022-12-21:
> Thats odd, i have not gotten an error like this maybe your sd card for flipper? Try rebooting your flipper

Tried reboot, tried reformating the SD. But the card is not the issue.
The problem also persists when I use the internal storage of the device.

xMasterX, 2022-12-21:
Bug can’t be reproduced, provide more details and check your hardware

closed 2022-12-21 by xMasterX