DarkFlippers/unleashed-firmware #227: Unleashed 022 transmission blocked on brutelrs

miodzicho, 2022-12-21:
### Describe the bug.

I tried to use brutelrs (eu option) from subghz remote, but seems tx is being blocked. 

### Reproduction

1. sub-ghz remote
2. fun_eu
3. arrow up - BruteLRS

RocketGod-git, 2022-12-21:
Please read this:
[DangerousSettings.md](https://github.com/DarkFlippers/unleashed-firmware/blob/dev/documentation/DangerousSettings.md)

miodzicho, 2022-12-21:
I seen it, but I'm concerden about point number 3 ... Extending frequency range and transmitting on frequencies that outside of hardware specs can damage your hardware!

RocketGod-git, 2022-12-21:
Yes, it's possible... but I haven't seen it happen yet. 

It's the only way it will work, unfortunately. Otherwise use another device designed for this frequency.

closed 2022-12-21 by miodzicho