DarkFlippers/unleashed-firmware #232: Transmission is blocked on frequency: 467750000

saormart, 2022-12-22:
### Describe the bug.

Hi there, it maybe not a bug, but something is wrong with my flipper.
On SubGhz I'm trying to "transmit" the following saved file: [Brute_first_12_loop.sub](https://github.com/RocketGod-git/Flipper_Zero/blob/main/subghz/Restaurant_Pagers/LRS_Pagers/Brute_first_12_loop.sub) and I get that flipper message: "Transmission is blocked" ...
This file transmits on this frequency: 467750000
And I have followed the process to install this firmware, but I have installed previous the Roguemaster fw.

### Reproduction

1. Followed the procedure of a fresh install of the Original/Official Firmware (SD-card Wiped)
2. Installed the Roguemaster FW by follow it's procedure on the roguemaster repository.
3. Didn't like the roguemaster, so I did the 1st step again to have a fresh install of the Original/Official Firmware (SD-card Wiped).
4. Followed the procedure to install this (unleashed-firmware) after have a fresh install of the Original/Official Firmware.
5. Tried to use transmit the: : [Brute_first_12_loop.sub](https://github.com/RocketGod-git/Flipper_Zero/blob/main/subghz/Restaurant_Pagers/LRS_Pagers/Brute_first_12_loop.sub) with no success.

### Target

Subghz

### Logs

```Text
N/A
```

### Anything else?

**Note:** I didn't tested on Roguemaster because I find out that file way after have this (unleashed-firmware) installed.
**Another important thing:** I notice that a feature from Roguemaster fw still working for the unleashed-firmware when I press and hold the middle button, it changes the flipper animation just like roguemaster! Not sure why or maybe it's saved on the flipper EEPROM that wasn't cleared during the flashing process... not sure...

xMasterX, 2022-12-22:
this is not roguemaster repo

closed 2022-12-22 by xMasterX

saormart, 2022-12-22:
I know this is not a roguemaster repo, but I currently have this unleashed-firmware from this repo installed, that's why I opened this ticket! 
I just mentioned that I had the roguemaster, but I'm no long using it, but this unleashed-firmware!
Can you at least try to test that file I mentioned please?
Thank you!

saormart, 2022-12-22:
> this is not roguemaster repo

I know this is not a roguemaster repo, but I currently have this unleashed-firmware from this repo installed, that's why I opened this ticket!
I just mentioned that I had the roguemaster, but I'm no long using it, but this unleashed-firmware!
Can you at least try to test that file I mentioned please?
Thank you!

xMasterX, 2022-12-22:
If you are using unleashed read this instructions 

https://github.com/DarkFlippers/unleashed-firmware/blob/dev/documentation/DangerousSettings.md

saormart, 2022-12-22:
> If you are using unleashed read this instructions
> 
> https://github.com/DarkFlippers/unleashed-firmware/blob/dev/documentation/DangerousSettings.md

I saw that documentation, but I didn't understand well...and I though that the unleashed would work, but after checking the CC1101 datasheet, now I understand why this procedure you mentioned has to be done to unlock those kind of frequency ranges!
I though that because it was in the spec range (300 to 928) would be fine, now I understand why and the risks!
Thank you!