DarkFlippers/unleashed-firmware #235: F6 hardware support

NetBUG, 2022-12-23:
### Description of the feature you're suggesting.

There are some dozens or even hundred of people owning Flipper with F6 hardware (who obtained it via hackathons, early tester program etc).

Is it hard to add target for F6 hardware? 

I checked https://github.com/flipperdevices/flipperzero-firmware/commit/c4a0847c99c68f6406c0f90f77df0191605f3207 carefully but TBH I feel confused why F6 became impossible to be supported

xMasterX, 2022-12-23:
Hundred of F6 flippers? Looks like you are wrong
There was only two events where F6 flippers was been, as far as i know
Also they are deprecated and will not be supported, so porting firmware to it doesn’t make sense

xMasterX, 2022-12-23:
Also, technically you can try to add support for F6
 
 But it will not be supported in unleashed or official firmware due to number of reasons
 
 So you can create your own fork that will have latest firmware with F6 hardware support

closed 2022-12-23 by xMasterX