DarkFlippers/unleashed-firmware #240: Picopass card

weeebdev, 2022-12-24:
### Describe the bug.

Hi! Recently I've been trying Unleashed Firmware. So, while browsing the apps, I launched PicoPass reader. And now it is just stuck at "Detecting picopass card". I cannot turn the flipper off, cuz none of the buttons work. Tried to connect via mobile app as well as via qFlipper (it says that it cannot connect). Nothing works. Can I somehow reset it? Or wait till battery will be drained? Thanks.

### Reproduction

1. Launch picopass reader
2. It's stuck!

### Logs

```Text
No logs, cuz I cannot connect to it via apps.
```

H3rrOberst, 2022-12-24:
Force a reboot via arrow left & back button.

weeebdev, 2022-12-25:
> Force a reboot via arrow left & back button at the same time for 3 sec.

Thanks! Anyway, it should be solved, isn't it?

closed 2022-12-25 by weeebdev

chosenone333, 2022-12-25:
I try to get in sub ghz remote it says check configSent from my iPhone