DarkFlippers/unleashed-firmware #241: Option to NOT remove TX Region Restrictions.

yodaluca23, 2022-12-25:
### Describe the enhancement you're suggesting.

Option to NOT remove TX Region Restrictions maybe something like what [Rouge Master](https://github.com/RogueMaster/flipperzero-firmware-wPlugins) Has...

### Anything else?

I would really like to use Unleashed, but I don't want to worry about accidentally/mistakenly using frequencies restricted from consumers...

xMasterX, 2022-12-26:
Why? Do you have basic understanding about TX power that flipper has?
I don't see any reason to lock frequencies in custom firmware, use official one if you don’t feel comfortable with region unlock

closed 2022-12-26 by xMasterX

mantacid, 2024-02-06:
maybe not removing them, but making the user confirm the use of a frequency that would normally be region-locked would be a good idea.

Grantie, 2024-03-17:
This would be an amazing addition. It scares me that I could be doing illegal things, so I'm using the official software. I would switch back to Unleashed if this is added.