DarkFlippers/unleashed-firmware #245: Backlight not immediately turning off when locking with a long press of up button

Salamafet, 2022-12-28:
### Describe the bug.

When locking the Flipper with the lock option in the Up menu, the backlight turn immediately off.

When using the long press Up button, the flipper lock but the backlight stay ON for the duration set in settings.

### Reproduction

1. Go to the home screen of the flipper
2. Long press the up button
Flipper lock but backlight not immediately turning off

### Anything else?

Not tested with the official firmware.

bunk3r, 2022-12-28:
You're right: when locked from the menu, system locks and the backlight goes away. 
When you long press the up button, flipper locks and the backlight turns off, but immediately returns on until the next timeout. (dev build 205 here)

Salamafet, 2023-01-07:
Tested today on the official firmware. 
Result: This function does not exist on the official firmware.

Suvero, 2023-02-07:
When it will be fixed ?

Salamafet, 2023-05-02:
@Suvero temporary fix in the last version
https://github.com/DarkFlippers/unleashed-firmware/releases/tag/unlshd-047

closed 2023-07-21 by xMasterX