DarkFlippers/unleashed-firmware #250: Marauder app not working

sadovsf, 2022-12-30:
### Describe the bug.

Seems that there is some issue with marauder GPIO wifi app. I just flashed FW into flipper wifi module. Everything went ok. So i connected board to flipper and find that app does not communicate properly with it. One of examples is log just flying by with readable text (seems communication is working properly) but app fails to read data from it.

Also when you close app. Flipper crashes on null pointer access (even if you just open and close it right away).

### Reproduction

1. Connect official Wifi board to Flipper
2. Open ESP32 Marauder app from GPIO folder on Flipper
3. Open log
4. (to see crash, close whole app using back button.)

### Target

unlshd-022

### Logs

```Text
Marauder flashing log:

You have chosen to flash Marauder on a WiFi devboard or ESP32-S2
You are most likely using a Flipper Zero WiFi Devboard or an ESP32-S2
Erasing firmware...
esptool.py v4.4
Serial port /dev/cu.usbmodem01
Connecting...
Chip is ESP32-S2 (revision v0.0)
Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 60:55:f9:ec:85:3c
Uploading stub...
Running stub...
Stub running...
Erasing region (may be slow depending on size)...
Erase completed successfully in 0.3 seconds.
Staying in bootloader.
Firmware erased!
Waiting 5 seconds...
Flashing ESP32 Marauder Firmware on a WiFi Devboard or ESP32-S2...
esptool.py v4.4
Serial port /dev/cu.usbmodem01
Connecting...
Chip is ESP32-S2 (revision v0.0)
Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 60:55:f9:ec:85:3c
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00004fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x00010fff...
Compressed 13648 bytes to 9382...
Wrote 13648 bytes (9382 compressed) at 0x00001000 in 0.3 seconds (effective 373.6 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.1 seconds (effective 356.0 kbit/s)...
Hash of data verified.
Compressed 212 bytes to 188...
Wrote 212 bytes (188 compressed) at 0x00010000 in 0.1 seconds (effective 26.2 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.
ESP32-S2 has been flashed with Marauder!
```

### Anything else?

My humble guess is that app is no longer up to date with either marauder FW or Unleashed Flipper FW.

xMasterX, 2022-12-30:
Issue can't be replicated, provide more details, also do you know that you need to reboot flipper after inserting the board?

closed 2022-12-30 by xMasterX

sadovsf, 2022-12-31:
Iam aware. Or rather I tried that as well. Is there something specific I could provide? 🤔

sadovsf, 2023-01-02:
OK, no idea what changed. I tried to redownload marauder FW and reflash it again (4th time already). This time it worked though and everything is OK 🤷‍♂️

Iam sorry for bothering, i did not do anything new this time but this time it simply worked 🤷‍♂️

sysfu, 2023-04-24:
For those struggling to get their WiFi Dev Boards flashed with the Marauder firmware I highly recommend this app https://github.com/SkeletonMan03/FZEasyMarauderFlash