DarkFlippers/unleashed-firmware #257: NFC stuck in emulation

hexan86, 2023-01-01:
### Describe the bug.

After emulating the UID of some saved tags, I've noticed that some in particular do not let you exit from the emulation and you have to press left+back.

Other tags work fine, Flipper get stuck emulating the UID of price tags.

Info about:
Unknown ISO tag
ISO 14443-4 (NFC-A)
UID: 1D 24 13 XX XX XX XX
ATQA: 00 44 SAK: 20

### Reproduction

1. Go to NFC menu
2. Select the tag to emulate
3. Press "Emulate UID"
4. Back doesn't work anymore

MichaelK-F, 2023-04-24:
I have had the same issue with my android (Samsung A53 5G) phone when using an app for writing to nfc tags then reading with a flipper. After emulation the flipper freezes up.

MiddleMan5, 2023-05-17:
I've noticed a similar issue on unlshd-048. Emulating a device from a saved nfc file allows me to exit normally with the back button until I place the eumlating flipper on a reader. The emulated tag is read correctly by the reader, but then I'm unable to exit with the back button and must reset

z3razerviper, 2023-08-07:
This is happening all the time for me now with amiibo......

GoldSilverC, 2023-10-03:
Still broken with amiibo especially

z3razerviper, 2023-10-03:
Yea its just gotten worse for me.

z3razerviper, 2023-10-12:
Still broken in 64

rekonnected, 2023-11-20:
Still broken in 65. This is quite frustrating because during the setup of some amiibo, you have to scan it three times just to get everything registered and configured fully. If any ONE of those fails, the process restarts and you need to hard boot the Flipper.

Would be nice to get some kind of acknowledgement of this issue, and ideally the removal of the Duplicate tag since nobody else seems to be talking about it anywhere else. No duplicate issue is referenced. Please prove its a dupe before labelling it as such @xMasterX

amec0e, 2023-11-23:
I had the freeze issue in 065 too and i have two solutions:

1. Install the latest dev build, this fixed it for me

2. Downgrade to 054 as the problem is not present in this version (at least for me this worked flawlessly)

**NOTE:**

If you choose method 1. **You must unlock your amiibo's** before they will be recognised by your switch, **otherwise you will get "not an amiibo"** error message.

To do this, scan your figurine > save > view saved file > unlock > unlock with reader > scan switch attempting to read amiibo > continue > scan your figurine and then save. 

Your .nfc file will now have the password on page 133 and the pack on page 134 which is required for the amiibo to be detected.

xMasterX, 2024-06-14:
Should be fixed now since nfc was fully refactored

closed 2024-06-14 by xMasterX