DarkFlippers/unleashed-firmware #263: Toniebox

waterguytony, 2023-01-05:
### Description of the feature you're suggesting.

Looking to see if it could be added to the NFC for the option to capture and store the data to emulate the Tonie box characters. The Xtreme has it but hands down prefer unleashed.

xMasterX, 2023-01-06:
xtreme has it because its based on RM, and RM merged this unfinished PR from OFW
https://github.com/flipperdevices/flipperzero-firmware/pull/1991

I don’t want to include unstable or unfinished features, so please wait until author finish his work on this PR, and it will be merged in OFW and here

waterguytony, 2023-01-06:
I will happily wait as seems unleashed is the only true stable one that actually allows use of the flipper.

supaeasy, 2023-07-18:
This works. NFC->Extra Actions ->Unlock SLIX-L ->Auth as TonieBox
Some Tonies have to be held sideways.

semaf, 2023-07-18:
Reading SLIX-L with the firmware of g3gg0 -> https://github.com/g3gg0/flipperzero-firmware.git works perfect.

jacobkjones, 2023-08-31:
Auth as Toniebox works, and then reading works, but for some reason, the Flipper doesn't save the file so that it could be emulated later.

supaeasy, 2023-10-05:
It works fine. I have over 20 Tonies stored and can emulate them.

jacobkjones, 2023-10-05:
I never figured out why my flipper doesn't want to save the Tonie files. Weird behavior.

supaeasy, 2023-10-05:
It does save them with cryptic names I have to say. But it saves them.

pentoo1989, 2023-10-20:
> It works fine. I have over 20 Tonies stored and can emulate them.

can you share them?

disintegr8te, 2023-12-11:
Somehow the Unblock Function for SLIX is gone in the newer VErsions.

closed 2023-12-11 by xMasterX

gotmygirls, 2024-01-02:
I have 47 Tonies so far saved on the flipper, would be nice if we could get a collection going. I am doing this strictly for my 2 year old daughter but the more we share the more we all have access to content wise for our children.

bsamuels453, 2024-01-02:
> I have 47 Tonies so far saved on the flipper, would be nice if we could get a collection going. I am doing this strictly for my 2 year old daughter but the more we share the more we all have access to content wise for our children.

please do! my wife is upset that some of the characters we've purchased don't have original voices, it would be amazing to be able to screen them in advance

supaeasy, 2024-01-02:
To my understanding emulating Tonies is a last resort security backup when we are on vacation and I forgot THE ONE AND ONLY FAVORITE (that seems to change weekly) for my kid. Do you really want to keep your Flipper in Kid's reach at all times? It is way more convenient to just buy a couple of creative tonies and update them with whatever is needed, when it is needed. I we wouldn't get so many Tonies as gifts we would only have these.

supaeasy, 2024-01-02:
> please do! my wife is upset that some of the characters we've purchased don't have original voices, it would be amazing to be able to screen them in advance

Yeah, new Arielle is shit.

gotmygirls, 2024-01-02:
₭Ｅ𝕖ＦƳŞ ¢𝓻𝔢Ⓐ𝕥ίόภ'𝔰#7687 is my discord not sure how we would get this started. I do like having a backup as she hides her favorite tonies often. I also like the idea of being able to "try before we buy" like the toy story and ALOT of others the voices can be that of different actors that in the movies. I have been hunting for both of the trolls but seem to only be able to source them on eBay. I don't want her having access to my flipper all the time but as a just in case with as small as those nfc files are you could store 100 of these tonies and still prob be under a full megabyte lol.[image]

D4N2135, 2024-01-02:
Have you shared these somewhere ? I'd like to have them for my two kids to listen to if you don't mind sharing them ?

gotmygirls, 2024-01-02:
> Have anyone shared these on here or somewhere else ? I'd like to have them for my two kids to listen to if you don't mind sharing them ?

I have searched since I got my youngest the tonie box over a year ago. I have not been able to find any reference on the web of anyone sharing the nfc files to the tonies they have. Make sure if you do find a way to source tonies you select lock tonie to household so it stays in your library. [image]

Tonies4theKids, 2024-01-02:
Is there others who are willing to contribute to this effort? If so are we to host on here or?

jacobkjones, 2024-01-03:
@Tonies4theKids I'd love to join in!

bsamuels453, 2024-01-03:
@Tonies4theKids i'm down to contribute the couple we've got

Tonies4theKids, 2024-01-03:
I think Private would be the only way going forward to do this. I had got so excited yesterday I uploaded alot and then realized that anyone would have access and only way to delete repo was to delete my account. Let me know what you guys wanna do and we can go that route. Even if it's a private group on say tele or something as encryption is a must. I only believe there a few of us out there who want to do this for our kids. Not that they need access to the flipper always but with the cost of these they add up and as a single dad I just want to know before I buy it that it will be a "favorite" of hers.  When she see's my flipper she instantly says piggy lol cause of Peppa.

lambyp, 2024-01-03:
I’ve got about 7 characters scanned. But have access to around 20. I’ve made a private group https://t.me/+TeMQMkx03zBkNDc0. Not sure if it’s worth it. I think we need to lock tonie to your household before you share any files. Also should we delete the comments in this thread?