DarkFlippers/unleashed-firmware #266: Auto Lock functionality

MMVipper, 2023-01-05:
### Description of the feature you're suggesting.

autolock can we have a way to have it "Lock With Pin" ? auto after 10-15-20-30-90sec?

H3rrOberst, 2023-01-06:
Go to Settings->Desktop->Auto-lock Time

Salamafet, 2023-01-07:
For your information, even if your Flipper is locked, it can be restarted with left and back button. 
When the Flipper start, the screen is unlocked by default.

The lock function is not a function for protecting the Flipper against thief or other joker (it can be easily reset). This function prevents the Flipper to make anything when it is in your pocket.
The pin locking it's against child, I suppose.

xMasterX, 2023-01-07:
> ### Description of the feature you're suggesting.
> autolock can we have a way to have it "Lock With Pin" ? auto after 10-15-20-30-90sec?
> 
> ### Anything else?
> _No response_

Please read @Salamafet response
His message describes all my thoughts about this, auto lock with pin is useless because it can be easily reset with two buttons, we had that function in the past, and it was removed because of that

closed 2023-01-07 by xMasterX

ClaraCrazy, 2023-01-08:
Sorry to just join in on that like that but why does your FW not lock after a (re)boot?