DarkFlippers/unleashed-firmware #269: KeeLoq: Supported manufacturers

Stichoza, 2023-01-06:
### Describe the enhancement you're suggesting.

Hello,

I'd love to help with improving support of additional KeeLoq manufacturers.

I have multiple `.sub` files recorded from Read mode with flipper (KeeLoq 64bit with key, fix, hop and button code, but without count and manufacturer).

Will it help the development process if I provide those sub files? Or anything else I can do?

P.S. Huge thanks for developing this firmware, it's the best of all FWs!

xMasterX, 2023-01-07:
Screenshot you provided is not related to our firmware in any case, this is someone’s commercial project and it says about his support, not ours

Attach your sub files here ill take a look on them

Stichoza, 2023-01-23:
Yes, I know that's a totally different thing on the screenshot. 

### Recorded signals
Here is a link to a gist containing captured signals: https://gist.github.com/Stichoza/fc61b41a4c5ae49c1fe59effd78c9713

### Remote
Remote has three buttons and looks like this: 
[image]

LY2NEO, 2023-01-26:
... just want to add, that some manufacturer keys which are already on flipper are partially decrypting where decrypted part of serial number are zeros, that gives oportunity to decrypt and see counter, but in revers when encrypting plain we getting wrong chipper and generated next hop is wrong and not working... I think it is due to not play everybody with neighbor garage doors 😀 but with power of math and powerful GPU I thing it is possible to get rigth ones 😉

calb33, 2023-03-18:
@xMasterX Just wondering if I can bother you with this one? It's a Keeloq remote called PROKEY2 from Australian company called CSD that is not detected yet on Unleashed FW. It runs on 433.92 and 650AM (even though it reads a bit funny on Freq Analyzer, jumps around 433.89 - 433.91). I've attached the RAW Read (renamed the sub into txt) from 10 button presses and photo of the PCB inside the remote (no FCC label unfortunately). Thanks! 😸 
[image]

[RAW_ButtonB.txt](https://github.com/DarkFlippers/unleashed-firmware/files/11007214/RAW_ButtonB.txt)

tomjschwanke, 2023-07-31:
Hi, I'd also like to contribute some signals. I have a remote that decodes as "KL Unknown", the remote looks like a "Novoferm MNHS", but has a different branding on it ("Siebau", but seems to be just a branding perhaps)

xMasterX, 2024-06-14:
> Hi, I'd also like to contribute some signals. I have a remote that decodes as "KL Unknown", the remote looks like a "Novoferm MNHS", but has a different branding on it ("Siebau", but seems to be just a branding perhaps)

Can you share RAW recordings of this remote??