DarkFlippers/unleashed-firmware #284: RFID Fuzzer

kuznetsov-m, 2023-01-15:
### Describe the bug.

Flipper zero crashed (NULL pointer dereference) with RFID Fuzzer tool

```
firmware_branch: release-cfw
firmware_branch_num: 3470
firmware_build_date: 31-12-2022
firmware_commit: 5c36043d
firmware_commit_dirty: false
firmware_target: 7
firmware_version: unlshd-023
```

[dump-2023-01-15-19-13-39.txt](https://github.com/DarkFlippers/unleashed-firmware/files/10420139/dump-2023-01-15-19-13-39.txt)

### Reproduction

1. Open RFID Fuzzer (EM4100 default mode)
2. Select mode BF Customer ID
3. Click TD- (5 times) for time delay: 5
4. Skip 3-10 UIDs
5. Click `back` button 2 times
6. Flipper zero crashed

### Logs

```Text
29 [default] Binding on background is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
29 [default] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
30 [default] Binding on animation is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
34 [default] Binding on background is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
39 [default] Binding on background is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
39 [default] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
40 [default] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
40 [default] Binding on background is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
47 [default] Binding on background is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
47 [default] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
47 [default] Binding on indicator is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
47 [default] Binding on background is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
353 [APP] qFlipper version 1.2.2 commit ba67025d 2022-12-08T19:31:11
354 [APP] OS info: macOS 11.6 11.6 20.6.0 Qt 6.3.1
381 [REG] Detected new device: VID_0x483:PID_0x5740
469 [DBG] Trying serial port  at /dev/cu.Bluetooth-Incoming-Port
470 [DBG] Trying serial port  at /dev/tty.Bluetooth-Incoming-Port
470 [DBG] Trying serial port  at /dev/cu.-CSRGAIA
470 [DBG] Trying serial port  at /dev/tty.-CSRGAIA
470 [DBG] Trying serial port  at /dev/cu.HUAWEIFreeBudsStudio-Se
470 [DBG] Trying serial port  at /dev/tty.HUAWEIFreeBudsStudio-Se
470 [DBG] Trying serial port  at /dev/cu.iPhone-WirelessiAP
470 [DBG] Trying serial port  at /dev/tty.iPhone-WirelessiAP
470 [DBG] Trying serial port  at /dev/cu.PSS-880-SerialPort
470 [DBG] Trying serial port  at /dev/tty.PSS-880-SerialPort
470 [DBG] Trying serial port  at /dev/cu.HUAWEIFreeBuds3i-Spp1
470 [DBG] Trying serial port  at /dev/tty.HUAWEIFreeBuds3i-Spp1
470 [DBG] Trying serial port flip_none ;) at /dev/cu.usbmodemflip_none ;)1
470 [DBG] Using  serial port flip_none ;) at /dev/cu.usbmodemflip_none ;)1
479 [RPC] Starting RPC session...
499 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
517 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
575 [RPC] RPC session started successfully.
575 [RPC] (1) System Protobuf Version START
631 [DBG] Detected protobuf version: 0.14
631 [RPC] (1) System Protobuf Version SUCCESS
631 [RPC] (2) Property Get START
743 [RPC] (2) Property Get SUCCESS
743 [RPC] (3) Storage Info @/ext START
795 [RPC] (3) Storage Info @/ext SUCCESS
795 [RPC] (4) Storage Stat @/ext/Manifest START
850 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
[... 378 lines cut ...]
74761 [RPC] RPC session started successfully.
74761 [RPC] (1) System Protobuf Version START
74819 [DBG] Detected protobuf version: 0.14
74819 [RPC] (1) System Protobuf Version SUCCESS
74820 [RPC] (2) Property Get START
74942 [RPC] (2) Property Get SUCCESS
74942 [RPC] (3) Storage Info @/ext START
74996 [RPC] (3) Storage Info @/ext SUCCESS
74996 [RPC] (4) Storage Stat @/ext/Manifest START
75067 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
75068 [RPC] (5) System Get DateTime START
75123 [DBG] Flipper time skew is -1271 milliseconds
75123 [RPC] (5) System Get DateTime SUCCESS
75123 [RPC] (6) System Set DateTime START
75183 [RPC] (6) System Set DateTime SUCCESS
75183 [RPC] Stopping RPC session...
75186 [RPC] RPC session stopped successfully.
75186 [REG] Registering the device
75186 [DEV] Version: unlshd-023 commit: 5c36043d radio: 1.13.3
75187 [BKD] Current device changed to "none ;)"
75288 [RPC] Starting RPC session...
75386 [RPC] RPC session started successfully.
75387 [RPC] (1) Gui Start ScreenStream START
75442 [RPC] (1) Gui Start ScreenStream SUCCESS
```

xMasterX, 2023-01-16:
Known issue, also logs from qflipper screen sharing will not help here at all, they only taken all space of this issue

also why your flipper name contains `;)` and space, this characters is not allowed for custom names
`none ;)`

kuznetsov-m, 2023-01-16:
I checked exist open issue before creation new one and I didn't find anything like it. If it duplicate - let's close it and continue discussion in original issue.
Sorry for long logs, just attach it for provide more information about Flipper state.

elrod16, 2023-01-18:
I've actually encountered this in other apps that I've launched from my favorites.

xMasterX, 2023-01-18:
> I've actually encountered this in other apps that I've launched from my favorites.

And? How its related to this issue?

elrod16, 2023-01-18:
I think they are related, I don't think it is specific to the SGhzBruteforce app, I think it deals with closing any app too fast.

xMasterX, 2023-01-18:
> I think they are related, I don't think it is specific to the RFID Bruteforce app, I think it deals with closing any app too fast.
> 
> Edit:typed wrong app name

This issue is known and present only with this plugin, tell what exact plugins have same issue?
Except ibutton fuzzer and rfid fuzzer

elrod16, 2023-01-18:
The infrared app, NFC Magic, Weather Station, for example can all be reliably crashed by backing out too fast.

xMasterX, 2023-01-18:
> The infrared app, NFC Magic, Weather Station, for example can all be reliably crashed by backing out too fast. Results in a furi_check error.

These all official plugins, we have not modified them and core parts of firmware also is not modified,
tell me your firmware version and exact steps to reproduce this bug, because i can't reproduce any crashes with just exiting that apps

elrod16, 2023-01-18:
I'm on Unleashed 23. For any of those apps, all it takes is entering them  interacting with a few menu items, then hammering the back button really fast. For infrared though, it seems to be connected to launching a remote directly instead of the main app and navigating to the saved remote.

xMasterX, 2023-07-21:
Rfid fuzzer issues has been fixed since app was rewritten from scratch

closed 2023-07-21 by xMasterX