DarkFlippers/unleashed-firmware #287: Docker

ErezBinyamin, 2023-01-17:
### Describe the enhancement you're suggesting.

It could be cool to maintain project dependencies in a Dockerfile. 

Some Benefits:
- Easier to get started with development
- Easier to build
- Good for automated tests

```Dockerfile
FROM ubuntu:20.04

# Update/Dependency
RUN apt-get update; \
	apt-get install -y \
		git \
		curl \
		wget

# Create non-root user: "builder"
RUN useradd -m builder && \
    adduser builder dialout
USER builder
WORKDIR /home/builder 

# Clone flipper firmware from GitHub
RUN git clone --recurse-submodules https://github.com/DarkFlippers/unleashed-firmware.git
WORKDIR unleashed-firmware
```

To run and develop interactivley
```bash
IMAGE_TAG=DarkFlippers/unleashed
PORT=/dev/$(shell ls -l /dev/serial/by-id/ | grep 'Flipper' | grep -o 'tty.*')
ls /dev/serial/by-id/ | grep -q 'Flipper' && docker run --device $(PORT):$(PORT) -it ${IMAGE_TAG} bash || docker run -it ${IMAGE_TAG} bash
```

For those who like to use Make in concert with Docker for automated dependency checking:
```Makefile
IMAGE_TAG=local/unleashed_firmware
CUSTOM_FLIPPER_NAME=myflipper
PORT=/dev/$(shell ls -l /dev/serial/by-id/ | grep 'Flipper' | grep -o 'tty.*')

.PHONY: all
all: .image_build

build: .image_build
	docker run --device $(PORT):$(PORT) ${IMAGE_TAG} CUSTOM_FLIPPER_NAME=${FLIPPER_NAME} ./fbt COMPACT=1 DEBUG=0 FORCE=1 flash_usb_full

dev:
	docker run --device $(PORT):$(PORT) -it ${IMAGE_TAG} bash

# Build Docker image
.image_build: Dockerfile
	docker build --tag ${IMAGE_TAG} .
	touch .image_build

clean:
	rm -f .image_build
	docker image rm ${IMAGE_TAG}
```

pdostal, 2023-01-19:
This maybe might be pull request?

ErezBinyamin, 2023-03-18:
It now is! And I made a replica issue. PR #392  and Issue: #393

derskythe, 2024-07-28:
I think better to use `cmake` instead of `make`.