DarkFlippers/unleashed-firmware #299: Somfy Smoove support

Kira-stuff, 2023-01-20:
### Description of the feature you're suggesting.

Hello,

I saw that support for Somfy telis and keytis have been added. Do you know if it's possible to add support for Somfy Smoove Remote ? 

https://shop.somfy.be/befr/media/catalog/product/cache/e576c2d73384ece72d3ff6a2c4412693/s/o/somfy-smoove-origin-rts-pure.jpg

It would be really nice if the flipper can be able to deal with this Somfy remote 

Thanks

assasinfil, 2023-01-20:
Hi!
We need rav recordings from this remote.

Kira-stuff, 2023-01-20:
I have the raw recordings for the smove remote is that enough ?

assasinfil, 2023-01-20:
Attach them here or contact me in telegram chat, my nickname is the same there.

Kira-stuff, 2023-01-24:
I post the update here for historic, but it seems that the raw recordings that I've send are not well recorded (analyzed by assasinfil) 

If someone else have raw recordings of Somfy Smoove Remote feel free to share

closed 2023-02-13 by xMasterX