DarkFlippers/unleashed-firmware #309: Flipper Zero v0.67.3 running Unleashed Firmware allows attackers to cause a Denial of Service via RF Jammer

shipcod3, 2023-02-04:
### Describe the bug.

My flipper zero v0.67.3 running Unleashed Firmware allows any attackers nearby that performs an RF Jam to also cause my Flipper Zero to reboot.

[image]

I have attached the PoC video as an attachment.

### Reproduction

Steps to reproduce:
1. Use a RF Jammer or simply [EvilCrow RF](https://github.com/joelsernamoreno/EvilCrow-RF) which has a Jammer.
2. Setup the Jammer with Frequency 433.92Mhz and then start.
3.In your flipper go to Sub-Ghz > Frequency Analyzer -> then tune it to the frequency of 433.92Mhz (scan mode)
4. Wait for RAW data and it should reboot after.

### Anything else?

https://user-images.githubusercontent.com/3483615/216756938-710b368e-482f-4b32-9e69-8c5638c30a38.MOV

xMasterX, 2023-02-04:
Read description of this commit

https://github.com/DarkFlippers/unleashed-firmware/commit/5853e0bdd9c7ef76ac9ed9bb05b262bf9fc5b374

Detect raw will be removed as soon as Bin Raw will be merged in OFW

xMasterX, 2023-02-04:
Also you running super old firmware, update it before creating issue

closed 2023-02-04 by xMasterX

shipcod3, 2023-02-05:
Understood