DarkFlippers/unleashed-firmware #319: password lock system

therberin, 2023-02-10:
### Description of the feature you're suggesting.

It would be good to implement security for the flipper owner

-to be able to turn off the flipper by holding the BACK button during the lock screen and when it starts, ask for the PIN again
(obviously more people have thought about that)

-be able to activate DUMMY MODE WITH PIN ask for the pin when returning to breiniack mode, and turn off in this mode
so when someone turns on the flipper it looks more like a tamagotchi and will be safe
or 
add a PIN to breiniak mode and ask for it every time you return from dummuy mode

I could have the same PIN or have another so that if you have a strong pin for the breiniak mode you can also blocked with another easy pin only for the poket 

- securyty options fot the START   start in dummy mode , start loked, start in last mode (may be not secure like now )

ClaraCrazy, 2023-02-10:
Gotta chime in real quick because people have suggested that many times on Unleashed, and other firmware too. Few things:

- Your sd card isn't encrypted. If I care about your data, I just take it. By the time you notice that I'm 30 miles away
- There is literally a reset code for the pin

<br><br>
- It's like adding a security code to your phone, in theory, it makes sense but everyone has it in their pocket at all times anyways, and the only one using it will be yourself, and everyone else will just bypass it within seconds

akopachov, 2023-03-09:
Just would like to explain what @ClaraCrazy said with one single meme image
[image]

Chuck3CZ, 2023-05-11:
@ClaraCrazy Yes, but if anybody ask, on security check for example: What is that, you can say Tamagochi for games or game console. 
Bcs dummy open only games.

But if they press Up -> Brainiac mode, there is no pin check. Anybody while borrowing for few seconds, under your semi-visual control, can get randomly to nfc/sub,... or saved data and they maybe dont know about bypassing password. 😇 Or if you wanna borrow it to kid for games only.

closed 2023-05-11 by therberin

ClaraCrazy, 2023-05-12:
> @ClaraCrazy Yes, but if anybody ask, on security check for example: What is that, you can say Tamagochi for games or game console. 
> Bcs dummy open only games.
> 
> But if they press Up -> Brainiac mode, there is no pin check. Anybody while borrowing for few seconds, under your semi-visual control, can get randomly to nfc/sub,...  they gonna know this is not only game console.
> 
> They maybe dont know about bypassing password. 😇 Or if you wanna borrow it to kid for games only.

They dont ask you. they fucking know. do you think they are low-iq morons? Everyone and their grandma knows what a flipper is. 

This is a pentesting device. You also shouldnt borrow it to kids, lol. Its not to play games.

I feel like this community is going downhill faster every single day

TheWebMachine, 2023-05-12:
Welcome to popularity! Not only does everyone and their grandma know about Flipper, now even Grandma is using one...and wants help from yooooouuuuuuu! 🤣