DarkFlippers/unleashed-firmware #329: Screen off option for standby.

ALD53105, 2023-02-12:
### Description of the feature you're suggesting.

It would be really cool to have an option to have a blank screensaver. I know it doesn’t use a lot of power, but it seems unnecessary to keep the screen running 24/7 if you don’t want the “Tamagotchi” experience.

YmartiY, 2023-02-12:
like the new screensaver :-)

YmartiY, 2023-02-12:
if you dont like the screensaver, hold the up button for three seconds...

YmartiY, 2023-02-12:
go to settings -> power -> power OFF... evtl. helps you out! flipper is booting fast.

ALD53105, 2023-02-12:
> go to settings -> power -> power OFF... evtl. helps you out! flipper is booting fast.

You missed the point here. I'm talking about in standby mode. I'm fully aware I can turn off the flipper, but I don't want to manually turn the device on and off several times a day.

YmartiY, 2023-02-12:
sorry, only a try to help you out for saving power...

valentinc03, 2023-02-27:
@ALD53105 hello I find it also interesting (but it has to be an option).

MichaelK-F, 2023-12-17:
I am a bit late to the discussion, but you could probably create a custom flipper animation with only one frame with nothing on it and just use that.

Dmitry422, 2024-09-18:
in this case will be better poweroff screen as blank screensaver - it will be more powersave.