DarkFlippers/unleashed-firmware #330: Update issue

KosteQ314, 2023-02-12:
So basically when I update to 027 all of the apps that I installed of of GitHub are question marks all with "API version mismatch" error. Anybody knows how to fix it?

DennisDeKoster, 2023-02-12:
I've had the same. Downloaded the Extra apps pack, but it didn't work. I found out that when I downloaded a separate file from GitHub, it does work. So I downloaded the zip file with the green button and everything works like a charm.

KosteQ314, 2023-02-12:
I tried redownloading the apps and installing the version with apps and without still the same problem for now I'm staying on 026

xMasterX, 2023-02-12:
> I tried redownloading the apps and installing the version with apps and without still the same problem for now I'm staying on 026

Remove apps folder and install latest build with extra pack included
Or remove old apps and install them manually

This is not an issue with update, this is issue with reading first line of changelog that tells you to update extra pack

closed 2023-02-12 by xMasterX