DarkFlippers/unleashed-firmware #341: Bluetooth BadUSB Feature

Gacut, 2023-02-15:
### Description of the feature you're suggesting.

It would be cool to get this feature from the video adapted into Unleashed firmware: 
[Link to Youtube Video](https://www.youtube.com/watch?v=lh99ssUy6FE)

Someone mentioned already that Bluetooth is not fully used it's potential on flipper zero so it could be a good start.

ClaraCrazy, 2023-02-16:
If people could wait, please? Its already been mentioned on the discord multiple times now.

I wrote this... my team wrote this. People instantly run to devs askint them to pull out ctrl-c ctrl-v. Theres an agreement to get it to unleashed eventually once it is bugfree and running fine, and once we had stable exclusivety of it for one version. 

Patience is a virtue

Gacut, 2023-02-16:
No need to be hostile about that. Not everyone is on every single discord server. I didn't see any mentions about this in here so i thought maybe this will start some kind of colaboration between devs.

Simple "It's in the works" would be enough.
Thanks for the info tho and sorry for asking.

ClaraCrazy, 2023-02-16:
After seeing other people take work withou lt crediting an estimated 3 seconds after release, my sympathy for people asking to integrate features from us is gone. Also, a collaboration, without involving me into the conversation by using a ping? Thats for you for next time, ping them. Looks like a silent `sweeping under the rug` otherwise. 

Having said that, sorry for the shitty response. Like i said, you can go say thanks to the "devs" that cant follow codex. Have a great day.

mayday77, 2023-02-16:
unf unf

Gacut, 2023-02-16:
> After seeing other people take work without crediting, an estimated 3 seconds after release, my sympathy for people asking to integrate features from us is gone. Also, a collaboration, without involving me into the conversation by using a ping? Thats for you for next time, ping them. Looks like a silent sweeping under the rug otherwise.

Having said that, sorry for the shitty response. Like i said, you can go say thanks to the "devs" that cant follow codex. Have a great day.

No hard feelings. And yeah, getting a credit in internet is a bitch. Have a nice day aswell :)

closed 2023-02-16 by Gacut

kimusan, 2023-02-16:
@ClaraCrazy what discord server is that? Would be nice to be able to follow such conversations. 
On a side note: I am fully for the whole "move to unleased eventually" agreement, but do remember that you have released FlipperExtreme under GPL3 license and shared the code publicly so there is nothing preventing people from copying it fully legally and use it in their non-commercial projects.

ClaraCrazy, 2023-02-16:
there is. and nano / xMasterX knows that :) 

and the server is on my repo