DarkFlippers/unleashed-firmware #345: [NFC] Mifare Classic 1k - Cant read all sector with all keys available

UrSuLa360, 2023-02-17:
### Describe the bug.

Mifare classic 1k tag with only 5 custom B keys. (5 sectors protected )
I know these keys, they are already added in the user key dictionary.
When reading the tag, the flipper got all keys but can't unlock the protected sectors.

[image]

### Reproduction

NFC->Read
Get flipper near tag

Wait for it :)

### Target

NFC

xMasterX, 2023-02-19:
Since we have no correct dumps of your card and can't replicate this issue what you need to do

Install latest official firmware dev branch, verify that your keys still in user dict (they should stay)
Then try to read it again, if it still does same thing

it can mean two problems, first, card can have access bits set in a way you cannot read those sectors, only write
for example system that uses this card writes different settings to that sectors, reads them, writes new data, then writes settings to lock them back

or if you think that issue is related to firmware and it will be present in OFW, create issue in official repo

UrSuLa360, 2023-02-19:
I have checked latest official firmware can't read the card anyway. 

But using and old official firmware can read it (i.e 72.1)
[image]

I will open this bug in official repo. 

Thank you very much for all the work you all are doing.

UrSuLa360, 2023-03-09:
Fixed in unlshd-034

Thank you all

closed 2023-03-09 by UrSuLa360