DarkFlippers/unleashed-firmware #346: can this be added

Louis-Beke, 2023-02-17:
### Description of the feature you're suggesting.

https://www.youtube.com/watch?v=lh99ssUy6FE

Wireless BadUSB with Flipper Zero's Inbuilt Bluetooth!

ClaraCrazy, 2023-02-17:
negative.

BBoyzthehacker, 2023-02-19:
why not clara?

Louis-Beke, 2023-02-19:
Hello,
Thanks for your email.
We will answer you when posible.
Best,
LSystemus Team

Nothankyou111, 2023-02-22:
Seriously lady what is your problem? Stop being a bitch

ezpzX1, 2023-02-22:
who where you talking to me or clara

ezpzX1, 2023-02-22:
why where you talkin to? them or clara

Nothankyou111, 2023-02-22:
Clara

Louis-Beke, 2023-03-03:
Oh sad

ClaraCrazy, 2023-03-03:
I wasnt serious, dw. Just love triggering kids. 

As mentioned on both our discord and the unleashed discord, it will be merged once its stable

BBoyzthehacker, 2023-03-03:
nice!

Louis-Beke, 2023-03-03:
Hello,
Thanks for your email.
We will answer you if possible.
Best,
LSystemus team

Nothankyou111, 2023-03-03:
> I wasnt serious, dw. Just love triggering kids.
> 
> As mentioned on both our discord and the unleashed discord, it will be merged once its stable

Just love being a cunt, ehh?

ClaraCrazy, 2023-03-04:
hell yea

BBoyzthehacker, 2023-03-04:
same tbh clara also when u gon update xtream it been 3 weeks

alexl83, 2023-05-09:
May I suggest to call it “BadBlue” when it gets launched in UNLSHD?

Louis-Beke, 2023-05-09:
yes

xMasterX, 2023-05-22:
done

closed 2023-05-22 by xMasterX