DarkFlippers/unleashed-firmware #356: Battery charging issue

Milkmik83, 2023-02-21:
### Describe the bug.

[image]
Version 030. Trying to charge Flipper when battery residual charge is 12% but the LED that usually was turning red while charging stays OFF.
It is not a cable issue since in Settings -> Power -> Battery Info it show that battery is already charged.

### Reproduction

With version 030 installed try to charge the battery

Milkmik83, 2023-02-21:
[image]
I've downgrade the FW to 022 to charge the battery.

xMasterX, 2023-02-21:
Issue can't be replicated, also no one reported any issues with charging, also nothing was changed in power management code

Looks like you having hardware issues

closed 2023-02-21 by xMasterX